flukejones/rog-core #2: Asus TUF505du laptop info

coolreader18, 2020-04-24:
I tried running it just as-is, and I got an error when starting the daemon. journalctl shows:
```
Apr 24 00:49:52 archlinux systemd[1]: rog-core.service: Start request repeated too quickly.
Apr 24 00:49:52 archlinux systemd[1]: rog-core.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- The unit rog-core.service has entered the 'failed' state with result 'exit-code'.
Apr 24 00:49:52 archlinux systemd[1]: Failed to start ROG Core Daemon.
-- Subject: A start job for unit rog-core.service has failed
-- Defined-By: systemd
-- Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- A start job for unit rog-core.service has finished with a failure.
-- 
-- The job identifier is 5702 and the job result is failed.
```

`lsusb` doesn't show anything with an id of `0b05:*`, nor even anything with AsusTek in the name. Seems like `FX*` models use a very different system from what's here :(. I do have the product info though, not that it would be much use:
```
❯ cat /sys/class/dmi/id/{product_{name,family},board_name}                        
TUF Gaming FX505DU_TUF505DU
TUF Gaming
FX505DU
```

flukejones, 2020-04-25:
What is the lsusb output?

coolreader18, 2020-04-26:
Just:
```
❯ lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 13d3:56a2 IMC Networks USB2.0 HD UVC WebCam
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 13d3:3548 IMC Networks Bluetooth Radio 
Bus 003 Device 008: ID 062a:4102 MosArt Semiconductor Corp. 2.4G Wireless Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

flukejones, 2020-04-27:
That's unusual.. What about a `dmesg |grep ASUS`?

My lsusb shows as:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0aaa Intel Corp. 
Bus 001 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

N-KEY Device being the keyboard. 

Or try [`usbhid-dump`](https://github.com/DIGImend/usbhid-dump) and see if you can get the info that way?

coolreader18, 2020-04-29:
Nope, no dice.
```
❯ dmesg | grep -i ASUS
[237614.977861] asus_wmi: Unknown key 33 pressed
```
```
❯ sudo usbhid-dump                   
003:009:000:DESCRIPTOR         1588185830.754717
 05 01 09 02 A1 01 85 03 09 01 A1 00 05 09 19 01
 29 05 15 00 25 01 75 01 95 05 81 02 75 03 95 01
 81 01 05 01 09 30 09 31 16 01 F8 26 FF 07 75 0C
 95 02 81 06 09 38 15 81 25 7F 75 08 95 01 81 06
 C0 05 0C 15 81 25 7F 75 08 95 01 0A 38 02 81 06
 C0 06 F2 F1 0A F2 F1 A1 01 85 05 09 00 95 06 75
 08 15 00 26 FF 00 91 02 09 00 95 05 75 08 15 00
 26 FF 00 81 02 C0
```
I tried doing `usbhid-dump --entity=all` and it only capture mouse events, not keyboard events. My guess is that this model hooks up the keyboard through something other than usb.

flukejones, 2020-04-30:
That discriptor is:

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x03,        //   Report ID (3)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x05,        //     Usage Maximum (0x05)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x05,        //     Report Count (5)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x03,        //     Report Size (3)
0x95, 0x01,        //     Report Count (1)
0x81, 0x01,        //     Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x16, 0x01, 0xF8,  //     Logical Minimum (-2047)
0x26, 0xFF, 0x07,  //     Logical Maximum (2047)
0x75, 0x0C,        //     Report Size (12)
0x95, 0x02,        //     Report Count (2)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x38,        //     Usage (Wheel)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x75, 0x08,        //     Report Size (8)
0x95, 0x01,        //     Report Count (1)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0x05, 0x0C,        //   Usage Page (Consumer)
0x15, 0x81,        //   Logical Minimum (-127)
0x25, 0x7F,        //   Logical Maximum (127)
0x75, 0x08,        //   Report Size (8)
0x95, 0x01,        //   Report Count (1)
0x0A, 0x38, 0x02,  //   Usage (AC Pan)
0x81, 0x06,        //   Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x06, 0xF2, 0xF1,  // Usage Page (Reserved 0xF1F2)
0x0A, 0xF2, 0xF1,  // Usage (0xF1F2)
0xA1, 0x01,        // Collection (Application)
0x85, 0x05,        //   Report ID (5)
0x09, 0x00,        //   Usage (0x00)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x09, 0x00,        //   Usage (0x00)
0x95, 0x05,        //   Report Count (5)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 118 bytes
```

[parser](https://eleccelerator.com/usbdescreqparser/). I'm guessing that yeah, it ain't hooked up via usb which makes it impossible to grab unfortunately.

closed 2020-04-30 by flukejones