flukejones/rog-core #3: Zephyrus G14

turowicz, 2020-04-30:
Any plans for supporting G14?

flukejones, 2020-05-01:
Daaaamn, that's a nice looking laptop. Keyboard looks similar, minus the home/end column of keys. I'm guessing the keyboard matrix for LEDs in gone, in place of the LED display on the top there.

What doesn't work on it? I doubt I'll be able to support the cover LEDs, but I could have a go at getting part of a lib sorted.

See [this section](https://github.com/flukejones/rog-core#other-laptops) of the readme - I'll need as much detail as I can get, but let's start with lsusb, and a `sudo usbhid-dump`. If those look promising then you'll have some work to do getting wireshark caps.

Eein, 2020-05-14:
Ive been looking into this a bit and my initial findings show:

1. Will need to wrap amd pstate (uses an amd 4900h(s) processor)
2. Most keys work on 5.6 latest with i believe the exception of screen brightness, rog, some multimedia keys and airplane mode.
3. Only the most expensive version has an led matrix

Ill work on collecting the required data soon when i have some free time but a couple of notes:

Ive had weird behavior/freezes on anything less than a 5.6 kernel, with the most success on 5.7rc5

Im on endeavorOS (arch) latest, but i have a coworker with the same laptop who wants to use popOS  so i can likely dual test 

Thanks for the cool package, i was going to start on something similar myself before i found this!

flukejones, 2020-05-14:
@Eein, is this specifically for the G14?

This app attempts to grab the USB device that the keyboard *usually* sits on, but apparently the G14 has a different setup here.

Eein, 2020-05-14:
Oh darn. Maybe i will have to do alot of work then. At least i have a reference haha.

Ill still give it a poke to gather some data and see if theres any similarities

dragonnn, 2020-05-15:
Nice that someone already have it. I will get my next week, hope w will get some functions in rog-core. I will try help too if I can.

flukejones, 2020-05-15:
[Parsing](https://eleccelerator.com/usbdescreqparser/) those HID descriptors:

```
0x06, 0x31, 0xFF,  // Usage Page (Vendor Defined 0xFF31)
0x09, 0x76,        // Usage (0x76)
0xA1, 0x01,        // Collection (Application)
0x85, 0x5A,        //   Report ID (90)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x05,        //   Report Count (5)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x00,        //   Feature (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x19, 0x00,        //   Usage Minimum (Unassigned)
0x2A, 0x3C, 0x02,  //   Usage Maximum (AC Format)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0x3C, 0x02,  //   Logical Maximum (572)
0x75, 0x10,        //   Report Size (16)
0x95, 0x02,        //   Report Count (2)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x06, 0x31, 0xFF,  // Usage Page (Vendor Defined 0xFF31)
0x09, 0x79,        // Usage (0x79)
0xA1, 0x01,        // Collection (Application)
0x85, 0x5D,        //   Report ID (93)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x1F,        //   Report Count (31)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0x91, 0x00,        //   Output (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x00,        //   Feature (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
0x06, 0x31, 0xFF,  // Usage Page (Vendor Defined 0xFF31)
0x09, 0x80,        // Usage (0x80)
0xA1, 0x01,        // Collection (Application)
0x85, 0x5E,        //   Report ID (94)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x05,        //   Report Count (5)
[... 108 lines cut ...]
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0xC0,        //   Report Count (-64)
0x75, 0x01,        //   Report Size (1)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x29, 0xEF,        //   Usage Maximum (0xEF)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 83 bytes
```

Vendor and product ID are `0b05:1866`, which matches the EC for the GX501 I have. So I'm wondering, how much of the app currently works?

> exception of screen brightness, rog, some multimedia keys and airplane mode.

This matches my experience. These keys are on a custom vendor page (as shown above in first descriptor block `0x85, 0x5A,        //   Report ID (90)`). This means they *should* already be getting grabbed by the app, and the correct keys emitted by the virtual device.

If someone can give a full report on which modes work on the keyboard, what model you have paired with if multi or matrix LED, then I can prepare some proper profiles hopefully including the AMD power stuff.

Screen backlight control should be handled by Gnome (or other) as the keycodes for that are emitted by the virtual device when those keys are pressed.

Looks like I might need to start a Wiki to help detail all this info better.

flukejones, 2020-05-15:
Also can someone clarify the model differences for me in regards to the top-cover led matrix, keyboard LEDS, and anything else that might help with toggling fan/power etc? I'm hoping the fan-mode stuff is the same as in the GX501 as it appears to be independent to CPU type.

Definitely curious for a list of LED modes that each model supports.

k-sap, 2020-05-16:
Definitely most of the G14 models do not have LED matrix on the lid.
I suppose power management, keyboard back-light, screen brightness control would be more important.

dragonnn, 2020-05-19:
Just got my G14 with 4700HS and 1660Ti. When I try run rog-core I got a SIGABORT:
`mateusz@rog ~/rog-core (master)> sudo /usr/bin/rog-core -d
[INFO ] Found GX502 or similar
[INFO ] Config loaded
[INFO ] INTERVAL: 1
[INFO ] MAX_PKT_SIZE: 64
[INFO ] SYNC: NoSync
[INFO ] TRANSFER_TYPE: Interrupt
[INFO ] ENDPOINT: 4
[INFO ] RogCore loaded
[WARN ] Fan mode: Fan mode not available
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ()', /home/mateusz/rog-core/rog-core/src/daemon.rs:56:5
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
fish: 'sudo /usr/bin/rog-core -d' zakończony sygnałem SIGABRT (Przerwij)
`
It is running on Arch Linux, what info can I provide more to testing it and getting it running.
When I install Windows back I can provide some Wiresharks dump

dragonnn, 2020-05-19:
Full backtrace:
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ()', /home/mateusz/rog-core/rog-core/src/daemon.rs:56:5
stack backtrace:
   0:     0x56075f4194fa - backtrace::backtrace::libunwind::trace::h84fcc841152fc264
                               at /build/rust/src/rustc-1.43.1-src/vendor/backtrace/src/backtrace/libunwind.rs:86
   1:     0x56075f4194fa - backtrace::backtrace::trace_unsynchronized::h9512da4a7923cf5a
                               at /build/rust/src/rustc-1.43.1-src/vendor/backtrace/src/backtrace/mod.rs:66
   2:     0x56075f4194fa - std::sys_common::backtrace::_print_fmt::h80a9838fbf6210c7
                               at src/libstd/sys_common/backtrace.rs:78
   3:     0x56075f4194fa - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hfbfe9ad65a0640ca
                               at src/libstd/sys_common/backtrace.rs:59
   4:     0x56075f38453c - core::fmt::write::hb409a30975c0f9f5
                               at src/libcore/fmt/mod.rs:1063
   5:     0x56075f418964 - std::io::Write::write_fmt::h7ee7c68a0296cf9a
                               at src/libstd/io/mod.rs:1426
   6:     0x56075f41856a - std::sys_common::backtrace::_print::h7cd9cb89129f3e0f
                               at src/libstd/sys_common/backtrace.rs:62
   7:     0x56075f41856a - std::sys_common::backtrace::print::ha9beca6e65f7f1cd
                               at src/libstd/sys_common/backtrace.rs:49
   8:     0x56075f41856a - std::panicking::default_hook::{{closure}}::h5e5a680f03f4adf9
                               at src/libstd/panicking.rs:204
   9:     0x56075f417c83 - std::panicking::default_hook::hb8f06365ed415b3d
                               at src/libstd/panicking.rs:224
  10:     0x56075f417c83 - std::panicking::rust_panic_with_hook::h1e03621439587007
                               at src/libstd/panicking.rs:470
  11:     0x56075f41786a - rust_begin_unwind
                               at src/libstd/panicking.rs:378
  12:     0x56075f382610 - core::panicking::panic_fmt::h2e4276b0cf7359c5
                               at src/libcore/panicking.rs:85
  13:     0x56075f3854e2 - core::option::expect_none_failed::h7e503ce831ca8ea3
                               at src/libcore/option.rs:1211
  14:     0x56075f403fee - <std::future::GenFuture<T> as core::future::future::Future>::poll::h6996eee008d3ca26
  15:     0x56075f3edcbd - tokio::runtime::handle::Handle::enter::hfa7927730acd5afa
  16:     0x56075f3f6b13 - rog_core::main::h50cdc698c5f13ea0
  17:     0x56075f3e9056 - std::rt::lang_start::{{closure}}::he588039299c62084
  18:     0x56075f3f6747 - main
  19:     0x7f7fd0f3c002 - __libc_start_main
  20:     0x56075f37809e - _start
  21:                0x0 - <unknown>

```

flukejones, 2020-05-19:
Hmm, thought I'd safely caught that. @dragonnn what kernel version are you running?

dragonnn, 2020-05-19:
@flukejones 
```
mateusz@rog ~/.c/autorun> uname -r
5.6.13-arch1-1
```

aspann, 2020-05-20:
asus-wmi patch: [asus-wmi-kernel-5.6.13.patch](https://lab.retarded.farm/zappel/zGentoo/-/raw/master/sys-power/nvidia-ratchet-g14/files/asus-wmi-kernel-5.6.13.patch)

tested using arch and gentoo. (use at own risk)

dragonnn, 2020-05-20:
With this path rog-core does boot up, and can control the keyboard backlight!

flukejones, 2020-05-22:
The `i8042` module issue may be because `rog-core` will unload any driver that has grabbed the interfaces it needs.

Can anyone please confirm the following? I'm having a hard time finding the clarity I need.

- Is there keyboard RGB?
- What modes does the keyboard backlight support?
- How much of `rog-core` is working for you? Are you using master branch built yourself?
- can anyone provide wireshark captures of the lid LED display? (follow directions in readme).

I'm attempting to add a profile for the G14. I've begun some of this work in master.

dragonnn, 2020-05-22:
As far I know the keyboard is on white backlight, and have 4 levels of light, it works fine with rog-core.
Switching fan modes, backlight keyboard, also all buttons on the keyboard are working too (rog button too). What can I try more? And yes, I am running master branch for now.
Unfortunately I don't have model with the animatrix

aspann, 2020-05-22:
The i8042 can't be used in "arch" out of thhe box, because it depends on PS/2 detection- so it works if it's builtin, but not as a module. (Tat is fixed by the patch from @dragonnn) 

My patches enables the detection of the WMI in general.

After those tweaks the following workes (for me at least):
- ROG Key
- Fan Profile
- Airplaine
- MediaKeys
- KB Backlight (4 levels)
- Brightness
- Touchpad toggle
- Sleep (hangs on suspend - but button technically works)

The matrixDisplay (on tha back) is single colored (white), as well as the Keybord backlight itself (also pure white).

tested under gentoo linux with v0.9.4 and v0.9.6 - as well as master.

dragonnn, 2020-05-22:
Yeah, I didn't have rog-core running when I was playing around with asus_nb_wmi module. The problem only exists when i8042 is build as a external module, it doesn't to load because they are no PS/2 devices on G14. And for G14 the asus_nb_wmi doesn't need i8042, it is only need for some ASUS Q500A

k-sap, 2020-05-22:
Did I understand correctly that with this repo power management on Asus G14 works good? I mean fans but also power draw on idle.

dragonnn, 2020-05-22:
For power drawn on idle you need enable Nvidia PRIME and set NVreg_DynamicPowerManagement to 0x02. More info here https://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/dynamicpowermanagement.html

flukejones, 2020-05-22:
> Did I understand correctly that with this repo power management on Asus G14 works good? I mean fans but also power draw on idle.

As @dragonnn writes above, you'll need to follow that for Optimus support. If the CPU were Intel, then along with fan modes you could set pstate settings like min/max freq and turbo-yes/no.

I don't have a G14 unfortunately, so someone will either need to write the code for AMD control, or provide me with some details. If it's a matter of writing a value to a path in `/sys` then it shouldn't be hard to support.

The pstate stuff is ignored if non-intel, so fan-modes should continue to just work.

If someone has the lid LED model, doing wireshark captures for that will be time-consuming. And quite likely dependent on if it's controlled via USB (seems highly likely).

aspann, 2020-05-22:
@flukejones I own that device with the aniMatrix (I'll investigate there later and hopefully provide you the needed information)

btw. The G14 series has only AMD APU(Vega).

And we (@dragonnn  and I) can confirm that the fanstates are working correctly. (using thermal-zone)

flukejones, 2020-05-22:
@aspann for the animatrix thingy I'm expecting something like [this](https://github.com/flukejones/rog-core/blob/master/wireshark_data/gx501/per_key_raw_bytes). But since it's a single colour it should be a lot simpler... maybe. I've no idea of the resolution of it. Maybe if we can get the corner LEDs byte sequence it'll give an idea of a pattern.

From memory, the packet sent to USB is 64 bytes, and at least 57 of those bytes are used for LED control (on the GX502). If the display is written by bit pattern, then it can control 456 of those pixels per packet - which means it's likely to be multiple packets. The GX502 uses ten packets to control the RGB...

There's likely to be an initialization byte sequence, and I suspect it will be part of the keyboard controller, which may make things a little easier.

dragonnn, 2020-05-22:
@flukejones acording to kernel docs boost on AMD is controled by /sys/devices/system/cpu/cpufreq/boost and I can confrim by looking at the turbostats and fans speed it works fine.

dragonnn, 2020-05-22:
Behind the way, did you consider add more profiles? I think it would be fine to have a silent profile with turbo enabled and a powersave with is also silent fan mode but with turbo disabled. Is this possible now?

Eein, 2020-05-22:
Wouldn't enabling and disabling turbo require the addition of a new pstate module in rust?
I think the current one is designed specifically for intel. 
(just something i noticed when i was digging around)

flukejones, 2020-05-22:
It will, but that's okay, I don't imagine it will be much different and I should be able to reuse part of the existing.Sent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/flukejones/rog-core/issues/3#issuecomment-632929172",
"url": "https://github.com/flukejones/rog-core/issues/3#issuecomment-632929172",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

flukejones, 2020-05-23:
> Behind the way, did you consider add more profiles? I think it would be fine to have a silent profile with turbo enabled and a powersave with is also silent fan mode but with turbo disabled. Is this possible now?

It's definitely possible, but it's not on my list of priorities sorry. I'll look at it some time - it would also be nice to get custom fan curves too, but this requires acpi calls which is a bit out of scope for now.

flukejones, 2020-05-23:
[New release for testing](https://github.com/flukejones/rog-core/releases/tag/v0.9.7)

KAMiKAZOW, 2020-05-26:
Would it be possible to work around missind PgUp/PgDn/Home/End keys by eg mapping them to Fn Left/Right and Shift+Fn Left/Right or so?

flukejones, 2020-05-26:
> Would it be possible to work around missind PgUp/PgDn/Home/End keys by eg mapping them to Fn Left/Right and Shift+Fn Left/Right or so?

If the Fn+Key combo actually emits anything sure, but it's very unlikely unless this is an actual built in function. Did this work in Windows?

KAMiKAZOW, 2020-05-26:
I don't own it (yet), so if someone else would be so kind to answer.

Alternatively, these are the key combos of ChromeOS: 

[image]

Eein, 2020-05-27:
> [New release for testing](https://github.com/flukejones/rog-core/releases/tag/v0.9.7)

My initial testing is that most things work except for the following (on GA401IV non-animatrix model):
I'll dive deeper into testing eventually - holding off on my RMA atm.

ROG key -> opens up displays settings in Gnome
Fan Key -> doesnt do anything afaik
Scissors key -> not sure what this is LOL

Wayland specific:
brightness controls do not work (will go like 1 shade darker but thats it)

Everything else seems to work perfectly! 

Thank you so much for all your hard work so far <3

flukejones, 2020-05-28:
> ROG key -> opens up displays settings in Gnome
This is intentional. I didn't really know what else to set the key to do. In future I will enable customization of it.

> Fan Key -> doesnt do anything afaik
Which kernel do you have? It should cycle the fan mode from <normal,boost,silent>, but there is no visual indicator of this unless you look at the logs or `cat /sys/devices/platform/asus-nb-wmi/throttle_thermal_policy`

> Scissors key
No idea either, what does it look like? If you can, press the key then look at the logs `systemctl status rog-core.service`, you should see something like:

```
May 28 08:45:20 zephyrus rog-core[32776]: [INFO ] Unmapped key, attempt passthrough: E9
May 28 08:45:20 zephyrus rog-core[32776]: [INFO ] Unmapped key, attempt passthrough: E9
```

From there I can map it to something.

> brightness controls do not work 
So that's an issue only with Wayland? `rog-core` just emits the two keys that Gnome and others interpret as bright up/down.

dragonnn, 2020-05-28:
On Xorg brightens control on G14 doesn't also work, it works after suspending and wakening up. Fan Key works. Scissors key I think on windows is used to start the Windows Screenshot Tool

flukejones, 2020-05-28:
> On Xorg brightens control on G14 doesn't also work, it works after suspending and wakening up. Fan Key works. Scissors key I think on windows is used to start the Windows Screenshot Tool

This seems unrelated to rog-core. Can you adjust brightness using the controls in your DE?

Per my previous reply, if you provide the output of that key I can map it to something if suggested.

dragonnn, 2020-05-28:
They keys works fine ;). It is something with the amdgpu driver

aspann, 2020-05-28:
Updated kernel-patches for G14: https://github.com/flukejones/rog-core/issues/3#issuecomment-631669269

Now includes k10temp patches for rernior as well.

k-sap, 2020-05-28:
Something like a touchpad feature (right click with a two finger tap) needs a different software solution?

flukejones, 2020-05-28:
> Something like a touchpad feature (right click with a two finger tap) needs a different software solution?

Yes. This is generally controlled by the touchpad driver or in your DE settings. If you use Gnome you can find it under "Mouse & Touchpad" >> "Tap to click"

aspann, 2020-05-30:
@k-sap if that doesn't work for you, you can adjust an override config for  synaptics if you're using Xorg:

/etc/X11/xorg.conf.d/99-synaptics-overrides.conf:
```
Section  "InputClass"
    Identifier  "touchpad overrides"
    MatchDriver  "synaptics"
    MatchIsTouchpad "on"

    # tap to click
    Option  "TapButton1"  "1"
    # two finger tap to rightclick
    Option  "TapButton2" "3"
    # vertical scroll
    Option  "VertEdgeScroll"  "1"
    # right click on lower-right corner
    Option  "RBCornerButton" "3"
EndSection
```

This works at least for my G14 very well. ;)

flukejones, 2020-06-25:
Are there any outstanding issues for G14 here?

jdeters, 2020-06-25:
If possible, it would be nice to get all of this working without kernel patches.

Also, if possible supporting charge_control_end_threshold under /sys/class/power_supply/BAT0 which was added in kernel 5.4 would be nice as well.

flukejones, 2020-06-25:
> If possible, it would be nice to get all of this working without kernel patches.

That would only be possible if those patches were upstreamed to the kernel proper. There's only so much that can be done in rog-core without the support.

> Also, if possible supporting charge_control_end_threshold under /sys/class/power_supply/BAT0 which was added in kernel 5.4 would be nice as well.

Wasn't aware of that! I'll add that in the next few days.

flukejones, 2020-06-26:
> Also, if possible supporting charge_control_end_threshold under /sys/class/power_supply/BAT0 which was added in kernel 5.4 would be nice as well.

Well... that was pretty quick to do. As of v0.12.0 you can set this using `rog-core -c 90` for example. Also reloaded on boot.

You do need to add below line 1 in `/etc/rogcore.conf`, `bat_charge_limit = 100` before reloading the new version.

jdeters, 2020-06-26:
I applied @aspann's asus-wmi kernel patch (which is also needed for this feature) and I can confirm that the battery threshold does work.

aspann, 2020-06-27:
@flukejones 
> Are there any outstanding issues for G14 here?

Yep, the threshold for the animatrix brightness level is still a todo. I just got my sensors and will do some measuerment next week.

@jdeters 
The asus-wmi patches are partial upstream since kernel 5.7.6. The asus-nb-wmi patch is still needed. (otherwise the DMI detection doesn't work). I already submitted it but didn't got a response from the maintainers by now.

flukejones, 2020-07-05:
It seems like most remaining issues here are now out of scope of this issue thread. The AniMe brightness thing is independent to all else here, and all else appears to require patches to kernels and those patches to be upstreamed.

Thanks to @everyone here for your help, patience, and testing :heart:

closed 2020-07-05 by flukejones