flukejones/rog-core #4: Zephyrus GM501GM support

wsay, 2020-05-01:
I have a Zephyrus GM501GM currently, which has a similar issue with the keyboard in Linux. I think it also has a very similar keyboard to the GX502.  More than happy to help with wireshark captures and whatever else you need, if you're open to it?

flukejones, 2020-05-02:
Looks similar. It's quite probable the GX502 profile will work.

Find the product ID and add it to this line https://github.com/flukejones/rog-core/blob/f12bb1d4f6ee8eeb8125e7f12b1858059a1609e8/rog-core/src/laptops.rs#L13

Otherwise [this section](https://github.com/flukejones/rog-core#other-laptops) of the readme should help with data collection.

If you want fancy LED effects (like per-key) then that might be a bit more involved but we can cut it down a little:
- boot Windows
- no usb devices attached
- open wireshark on USB hub (2 normally)
- open the Aura effect creator
- create a layout with a single key selected
- add an effect to that single key, one colour, shortest active time (I think .5s)
- press the "Play" icon
- stop and save the wireshark data...

But we can do this for one key in each row, say ROG, F4, 5, R, F, C, Left part of Space, and then the full number pad. Ideally I just want a file like [this](https://github.com/flukejones/rog-core/blob/master/wireshark_data/per_key_raw_bytes) that I can dump in LibreOffice Calc.

You can get the plain hex stream from selecting the relevant packet like:
[image]

Then right-click on "Leftover Capture Data" -> Copy -> ...as Hex-Stream, then has in file like the example above. Don't worry about comma-spacing unless you want to do it.

It's worth trying the example to see what happens: `cargo run --example per-key-effect-2`

wsay, 2020-05-04:
Okay, as far as I can tell the product ID is the second of the two already listed. Git-cloned the code and installed the libraries, then tried to make. On Pop!_OS 20.04 I had to also apt install libclang-dev before the compile would run succesfully.

After I got it compiled installed it, and tried some commands, but I see the error 'Error: "The name org,rogcore,Daemon was not provided by any .service files"'. Also when running rog-core -d (to see if this helped with the prior issue) got an error after "[INFO] RogCore loaded" of 'Error: Custom { kind: WriteZero, error: "failed to write whole buffer" }'.

Happy to try any debugging you can suggest, but I'm not a rust coder or all that familiar with Linux dev so wasn't sure where to go from there.

flukejones, 2020-05-04:
make should install the service files required - then you need to enable and start the service:
```
$ sudo systemctl start rog-core.service
$ sudo systemctl enable rog-core.service
```

But if there are issues with running with `rog-core -d` before that, then we need to debug it. There are new builds in the PPA if that helps.

wsay, 2020-05-04:
[hist.txt](https://github.com/flukejones/rog-core/files/4577459/hist.txt)
Attached a log of the commands, run just now.

flukejones, 2020-05-04:
Please use sudo to run in daemon mode. It requires root.

flukejones, 2020-05-05:
Oh sorry, it looks like you need to reboot to have the dbus rules kick in properly. Also check the existence of `/etc/dbus-1/system.d/rog-core.conf`

wsay, 2020-05-05:
Made sure to run with sudo, and that file exists with the contents: `<!DOCTYPE busconfig PUBLIC
          "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
          "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>
    <policy group="adm">
        <allow send_destination="org.rogcore.Daemon"/>
        <allow receive_sender="org.rogcore.Daemon"/>
    </policy>
    <policy group="sudo">
        <allow send_destination="org.rogcore.Daemon"/>
        <allow receive_sender="org.rogcore.Daemon"/>
    </policy>
    <policy user="root">
        <allow own="org.rogcore.Daemon"/>
        <allow send_destination="org.rogcore.Daemon"/>
        <allow receive_sender="org.rogcore.Daemon"/>
    </policy>
</busconfig>
` even after a reboot I still get the same error messages, tried both the version off the PPA and a from-source compilation again.

wsay, 2020-05-05:
sudo systemctl status rog-core reports: 

will@pop-os:/usr/lib/systemd/system$ sudo systemctl status rog-core
● rog-core.service - ROG Core Daemon
     Loaded: loaded (/lib/systemd/system/rog-core.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2020-05-05 10:10:45 BST; 4s ago
    Process: 4247 ExecStart=/usr/bin/rog-core -d (code=exited, status=1/FAILURE)
   Main PID: 4247 (code=exited, status=1/FAILURE)

May 05 10:10:45 pop-os systemd[1]: rog-core.service: Scheduled restart job, restart counter is at 5.
May 05 10:10:45 pop-os systemd[1]: Stopped ROG Core Daemon.
May 05 10:10:45 pop-os systemd[1]: rog-core.service: Start request repeated too quickly.
May 05 10:10:45 pop-os systemd[1]: rog-core.service: Failed with result 'exit-code'.
May 05 10:10:45 pop-os systemd[1]: Failed to start ROG Core Daemon.
will@pop-os:/usr/lib/systemd/system$

flukejones, 2020-05-05:
Please try `sudo journalctl -b -u rog-core.service`

wsay, 2020-05-05:
[rog-servicelog.txt](https://github.com/flukejones/rog-core/files/4580177/rog-servicelog.txt) File attached - I did a little bit of debugging and it seems to be the line `rogcore.fan_mode_reload(&mut config).await?; `that causes the issue. I've upgraded to kernel 5.6 as you mentioned some possible incompatibilities with fan modes and older kernels, but it does the same.

flukejones, 2020-05-05:
Looks like it's not writing to the path. Can you check if one of these paths exist?

```
static FAN_TYPE_1_PATH: &str = "/sys/devices/platform/asus-nb-wmi/throttle_thermal_policy";
static FAN_TYPE_2_PATH: &str = "/sys/devices/platform/asus-nb-wmi/fan_boost_mode";
```

But, looking at the error it seems that the path is found, but it fails to write for another reason.

Also need `uname -r`. I run 5.6.6, which is what i test with. 

Please try again with latest master, you need to do:

```
sudo make install && sudo systemctl daemon-reload && sudo systemctl restart rog-core.service
```

wsay, 2020-05-05:
fan_boost_mode exists - and yes it finds the file but cannot write to it. I also can't write anything to it using other tools without whatever program locking up (even as root). Uname -r gives me `5.6.7-050607-generic` - I will test now with the latest master.

wsay, 2020-05-05:
Testing with latest master:

[INFO ] Found GX502 or similar
[INFO ] Config loaded
[INFO ] INTERVAL: 1
[INFO ] MAX_PKT_SIZE: 64
[INFO ] SYNC: NoSync
[INFO ] TRANSFER_TYPE: Interrupt
[INFO ] ENDPOINT: 4
[INFO ] RogCore loaded
[ERROR] Could not write fan mode: Custom { kind: WriteZero, error: "failed to write whole buffer" }
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ()', /home/will/rog-core/rog-core/src/core.rs:100:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Aborted

flukejones, 2020-05-05:
`cat /etc/rog-core.conf`, if there is no `fan_mode = ` please add line `fan_mode = 0` to top of file. Won't hurt to paste content here, then delete that file before rerunning.

flukejones, 2020-05-05:
On my machine I have `throttle_thermal_policy`. Can you do a `/sys/devices/platform/asus-nb-wmi/fan_boost_mode`? We need to see what the contents are. Are there any other files in `/sys/devices/platform/asus-nb-wmi/`?

If we can't get it working I'll change the app to log an error and carry on rather than crash, then you can at least use the fan button to change the CPU pstates.

wsay, 2020-05-06:
A clone of the main repository appears to be working now - did you make another change? Or is it just my laptop being capricious.. in answer to questions:

/etc/rog-core.conf doesn't appear to exist.

`will@pop-os:/sys/devices/platform/asus-nb-wmi$ ls
cpufv   driver_override  hwmon  modalias  subsystem
driver  fan_boost_mode   input  power     uevent
will@pop-os:/sys/devices/platform/asus-nb-wmi$ cat fan_boost_mode
0
will@pop-os:/sys/devices/platform/asus-nb-wmi$ 
`

flukejones, 2020-05-06:
Can you check your logs for `Could not write to` please? If there is an error it should be appended to that.

:facepalm: `/etc/rogcore.conf` sorry. I really should have kept naming consistent with the crate.

flukejones, 2020-05-06:
For the fan stuff, it looks like it isn't the fault of the app at all for you. Can you try similar to the following:

```
[~/P…/J…/mobile_application] (milestone/firmware_api[?]) 
[luke@zephyrus]$ echo 0 | sudo tee /sys/devices/platform/asus-nb-wmi/throttle_thermal_policy 
[sudo] password for luke: 
0

[~/P…/J…/mobile_application] (milestone/firmware_api[?*]) 
[luke@zephyrus]$ cat /sys/devices/platform/asus-nb-wmi/throttle_thermal_policy 
0

[~/P…/J…/mobile_application] (milestone/firmware_api[?*]) 
[luke@zephyrus]$ echo 2 | sudo tee /sys/devices/platform/asus-nb-wmi/throttle_thermal_policy 
2

[~/P…/J…/mobile_application] (milestone/firmware_api[?*]) 
[luke@zephyrus]$ cat /sys/devices/platform/asus-nb-wmi/throttle_thermal_policy 
2
```

It might even be worth trying the 5.7rc kernels if you're using `ukuu` to help manage kernels.

I'm very curious to what happens when you run one of the example RGB effects now too:

```
cargo run --example iterate-keys
cargo run --example comet
cargo run --example pulser
cargo run --example ball
```

iterate-keys should go through all rows from left to right, top-down. I doubt the numpad will light up, but would like to see where the boundary is.

wsay, 2020-05-06:
[test-commands.txt](https://github.com/flukejones/rog-core/files/4586747/test-commands.txt)
Tried the first round of commands - file is attached. Seems to be possible to write to the file, and it sticks. But the program doing the writing locks up, and I have to ctrl+C out of it.

wsay, 2020-05-06:
Running iterate keys results in: 
   Compiling rog-aura v0.9.1 (/home/will/rog-core/aura)
    Finished dev [optimized] target(s) in 31.79s
     Running `target/debug/examples/iterate-keys`
GOT Message { Type: Signal, Path: "/org/rogcore/Daemon", Interface: "org.rogcore.Daemon", Member: "LedCancelEffect", Sender: ":1.19", Serial: 20, Args: [] }
^C
will@pop-os:~/rog-core$ 

All of the keys remain lit up - this may be because I this keyboard is lit in zones I think, rather than per-key.

flukejones, 2020-05-06:
Please follow the details [here](https://github.com/flukejones/rog-core#supporting-more-laptops) to provide me more information. 

I have added a multi-zone setup, but I can't test it, and I do need information about your machine to correctly set this up.

flukejones, 2020-05-08:
To be clear, what I need mostly is this:

---
lsusb:

First do `lsusb |grep 0b05` and check the part after `0b05:`, output looks like:

```text
Bus 001 Device 005: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
```

Then do `sudo lsusb -vd 0b05:1866 > ~/laptop_info` and give that to me.

Other helpful info can be gained from `sudo usbhid-dump`, for which you may need to unload kernel drivers. Please google
this.

Also required (for my book-keeping of data):
- `cat /sys/class/dmi/id/product_name`
- `cat /sys/class/dmi/id/product_family`
- `cat /sys/class/dmi/id/board_name`

---

usbhid-dump should work after running the rog-core in daemon mode then stopping it.

You don't need to worry about wireshark captures, but a list of modes your keyboard supports would be helpful.

wsay, 2020-05-08:
[lsusbinfo & sysclassname.txt](https://github.com/flukejones/rog-core/files/4601003/lsusbinfo.sysclassname.txt)
[usb-hid-dump.txt](https://github.com/flukejones/rog-core/files/4601006/usb-hid-dump.txt)
[zephyrusgm501_info.txt](https://github.com/flukejones/rog-core/files/4601008/zephyrusgm501_info.txt)

Attached three files with those outputs - I'll try and get a list of modes, don't have a windows install at the moment on this machine.

flukejones, 2020-05-09:
Everything I'm seeing and finding in search for information seems to indicate that this keyboard does have per-key LED. With daemon running, do you get a rainbow effect if you try: `rog-core led-mode rainbow` (this [video](https://www.youtube.com/watch?v=XmQaFey0lek) shows a rainbow mode).

The example special effects also require the daemon to be running.

wsay, 2020-05-10:
Maybe I'm mistaken then - apologies. Rainbow does indeed work - I worked through the list of effects you have:
stable - works
breathe - works
cycle - works
rainbow - works
rain - keyboard goes dark
disco - keyboard goes dark
highlight - keyboard goes dark
laser - keyboard goes dark
ripple - keyboard goes dark
pulse - works
thin-zoomy - keyboard goes dark
wide-zoomy - keyboard goes dark
multi-static - Only the numpad llights up, with a bit of bleed to the right edge of the keyboard but doesn't look like it's lit there, just leaking through.

From memory, these I think are similar modes to what I had available in Windows, along with some customisation options.

flukejones, 2020-05-10:
Highlight, laser, and ripple require pressing buttons to show effect.

All modes can have "--help" appended to show what the extra options are. Can you check those too please? And also confirm the examples don't work with the daemon running?

I'll begin building up a profile for this laptop using what you've provided.

wsay, 2020-05-10:
If it would be helpful I can do a video working through the commands and stick it on youtube? The effects I listed working all work, with all options seeming to work Ok. The ones listed as not working don't seem to differ when the other options are changed.
There's no clear zone boundaries in rainbow mode, but definitely only the numpad lights up in multi-static, colour changes correctly for that zone.

flukejones, 2020-05-22:
It might be helpful to get a video of multizone if possible (and rainbow). It would be great if you or someone else were able to provide wireshark captures of multizone modes in Windows. It looks like you can do a reasonable imitation of some other effects with it.

I've started trying to implement proper profiles for laptops, so cycling through modes using aura-keys on keyboard if they are there should work better. Otherwise the daemon should give feedback if you send an unsupported mode from the client.

flukejones, 2020-05-23:
[New release for testing](https://github.com/flukejones/rog-core/releases/tag/v0.9.7)

Please try `rog-core led-mode multi-static --help`

wsay, 2020-06-03:
Hi, am unable to test as I unfortunately don't have access to that laptop anymore! But was using it successfully with the prior version for some time. Hopefully someone else with a similar machine will come help out. Think this is a great project.

flukejones, 2020-07-05:
I'm going to close this as unable to test further for now. Feedback from others indicates that the multi-zone modes I've set up do work, so I'll work on adding the others.

closed 2020-07-05 by flukejones