flukejones/rog-core #6: Need to press fan mode button twice to change fan mode

dragonnn, 2020-05-21:
```
maj 21 13:25:59 rog rog-core[111668]: [INFO ] Current fan mode: Silent
maj 21 13:25:59 rog rog-core[111668]: [INFO ] Fan mode stepped to: Normal
maj 21 13:26:13 rog rog-core[111668]: [INFO ] Current fan mode: Silent
maj 21 13:26:13 rog rog-core[111668]: [INFO ] Fan mode stepped to: Normal
maj 21 13:26:26 rog rog-core[111668]: [INFO ] Current fan mode: Normal
maj 21 13:26:26 rog rog-core[111668]: [INFO ] Fan mode stepped to: Boost
maj 21 13:26:34 rog rog-core[111668]: [INFO ] Current fan mode: Normal
maj 21 13:26:34 rog rog-core[111668]: [INFO ] Fan mode stepped to: Boost
```
On G14 I need to press fan button mode twice to switch fan profile, it is also visible in logs, twice it goes to Normal, then twice to go to Boost.
I don't think this is actually linked to G14

aspann, 2020-05-21:
I can confirm that behaviour on the ROG Zephyrus G14.

flukejones, 2020-05-22:
Confirmed on GX501. Will fix soon.

flukejones, 2020-05-22:
Fixed in commit 5e781cbd3d8e1ef2c95ed270f03fb2f6a47f8d93

closed 2020-05-22 by flukejones

dragonnn, 2020-05-22:
Thanks, I can confirm it works now for me.

aspann, 2020-05-22:
I can also confirm the fix works. 

Are you planning a new release for that? (would be great, because I'm providing a ebuild for that - so also the gentoo user can use this tool)

Feel free to adopt my ebuild: [sys-power/rog-core](https://lab.retarded.farm/zappel/zGentoo/-/tree/master/sys-power/rog-core)

cheers.

flukejones, 2020-05-22:
> Are you planning a new release for that?

[https://github.com/flukejones/rog-core/releases/tag/v0.9.6](https://github.com/flukejones/rog-core/releases/tag/v0.9.6)
 
> Feel free to adopt my ebuild: [sys-power/rog-core](https://lab.retarded.farm/zappel/zGentoo/-/tree/master/sys-power/rog-core)

Let me think about it, I don't use Gentoo, and adopting your build would require me maintaining it, but I'll add it in to the readme.

aspann, 2020-05-22:
great, many thanks!