flukejones/rog-core #8: Randmon crash and doesn't start any more

dragonnn, 2020-05-22:
Super strange, it just stopped working and doesn't start any more even after full reboot. 
Crash log:
```
[INFO ] Found GX502 or similar
[INFO ] Config loaded
[INFO ] INTERVAL: 1
[INFO ] MAX_PKT_SIZE: 64
[INFO ] SYNC: NoSync
[INFO ] TRANSFER_TYPE: Interrupt
[INFO ] ENDPOINT: 4
[ERROR] Invalid parameter
thread 'main' panicked at 'Invalid parameter', <::std::macros::panic macros>:5:6
stack backtrace:
   0:     0x55b13b13ff9a - backtrace::backtrace::libunwind::trace::h84fcc841152fc264
                               at /build/rust/src/rustc-1.43.1-src/vendor/backtrace/src/backtrace/libunwind.rs:86
   1:     0x55b13b13ff9a - backtrace::backtrace::trace_unsynchronized::h9512da4a7923cf5a
                               at /build/rust/src/rustc-1.43.1-src/vendor/backtrace/src/backtrace/mod.rs:66
   2:     0x55b13b13ff9a - std::sys_common::backtrace::_print_fmt::h80a9838fbf6210c7
                               at src/libstd/sys_common/backtrace.rs:78
   3:     0x55b13b13ff9a - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hfbfe9ad65a0640ca
                               at src/libstd/sys_common/backtrace.rs:59
   4:     0x55b13b0abd1c - core::fmt::write::hb409a30975c0f9f5
                               at src/libcore/fmt/mod.rs:1063
   5:     0x55b13b13f404 - std::io::Write::write_fmt::h7ee7c68a0296cf9a
                               at src/libstd/io/mod.rs:1426
   6:     0x55b13b13f00a - std::sys_common::backtrace::_print::h7cd9cb89129f3e0f
                               at src/libstd/sys_common/backtrace.rs:62
   7:     0x55b13b13f00a - std::sys_common::backtrace::print::ha9beca6e65f7f1cd
                               at src/libstd/sys_common/backtrace.rs:49
   8:     0x55b13b13f00a - std::panicking::default_hook::{{closure}}::h5e5a680f03f4adf9
                               at src/libstd/panicking.rs:204
   9:     0x55b13b13e723 - std::panicking::default_hook::hb8f06365ed415b3d
                               at src/libstd/panicking.rs:224
  10:     0x55b13b13e723 - std::panicking::rust_panic_with_hook::h1e03621439587007
                               at src/libstd/panicking.rs:470
  11:     0x55b13b13e30a - rust_begin_unwind
                               at src/libstd/panicking.rs:378
  12:     0x55b13b13e2b0 - std::panicking::begin_panic_fmt::hc43e1369214a6caa
                               at src/libstd/panicking.rs:332
  13:     0x55b13b131d21 - daemon::daemon::start_daemon::{{closure}}::{{closure}}::he36b4981a748d04a
  14:     0x55b13b1312ae - <std::future::GenFuture<T> as core::future::future::Future>::poll::h9f92e186da602c5a
  15:     0x55b13b11b9fd - rog_core::main::h52183f91b7196cea
  16:     0x55b13b10f856 - std::rt::lang_start::{{closure}}::hb73d94e66ca68283
  17:     0x55b13b11b327 - main
  18:     0x7f9686b12002 - __libc_start_main
  19:     0x55b13b09f09e - _start
  20:                0x0 - <unknown>
fish: Job 2, 'sudo env RUST_BACKTRACE=full /u…' zakończony sygnałem SIGABRT (Przerwij)
```
Any idea what could cause this? Even reboot didn't help

flukejones, 2020-05-22:
Hmm, try removing `/etc/rogcore.conf`

dragonnn, 2020-05-22:
Nope, still the same error.
```
[INFO ] Found GA401 series
[INFO ] Config loaded
[INFO ] INTERVAL: 1
[INFO ] MAX_PKT_SIZE: 64
[INFO ] SYNC: NoSync
[INFO ] TRANSFER_TYPE: Interrupt
[INFO ] ENDPOINT: 4
[ERROR] Invalid parameter
thread 'main' panicked at 'Invalid parameter', <::std::macros::panic macros>:5:6
stack backtrace:
   0:     0x561753e1ac8a - backtrace::backtrace::libunwind::trace::h84fcc841152fc264
                               at /build/rust/src/rustc-1.43.1-src/vendor/backtrace/src/backtrace/libunwind.rs:86
   1:     0x561753e1ac8a - backtrace::backtrace::trace_unsynchronized::h9512da4a7923cf5a
                               at /build/rust/src/rustc-1.43.1-src/vendor/backtrace/src/backtrace/mod.rs:66
   2:     0x561753e1ac8a - std::sys_common::backtrace::_print_fmt::h80a9838fbf6210c7
                               at src/libstd/sys_common/backtrace.rs:78
   3:     0x561753e1ac8a - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hfbfe9ad65a0640ca
                               at src/libstd/sys_common/backtrace.rs:59
   4:     0x561753d97dfc - core::fmt::write::hb409a30975c0f9f5
                               at src/libcore/fmt/mod.rs:1063
   5:     0x561753e1a0f4 - std::io::Write::write_fmt::h7ee7c68a0296cf9a
                               at src/libstd/io/mod.rs:1426
   6:     0x561753e19cfa - std::sys_common::backtrace::_print::h7cd9cb89129f3e0f
                               at src/libstd/sys_common/backtrace.rs:62
   7:     0x561753e19cfa - std::sys_common::backtrace::print::ha9beca6e65f7f1cd
                               at src/libstd/sys_common/backtrace.rs:49
   8:     0x561753e19cfa - std::panicking::default_hook::{{closure}}::h5e5a680f03f4adf9
                               at src/libstd/panicking.rs:204
   9:     0x561753e19413 - std::panicking::default_hook::hb8f06365ed415b3d
                               at src/libstd/panicking.rs:224
  10:     0x561753e19413 - std::panicking::rust_panic_with_hook::h1e03621439587007
                               at src/libstd/panicking.rs:470
  11:     0x561753e18ffa - rust_begin_unwind
                               at src/libstd/panicking.rs:378
  12:     0x561753e18fa0 - std::panicking::begin_panic_fmt::hc43e1369214a6caa
                               at src/libstd/panicking.rs:332
  13:     0x561753d78ad1 - daemon::daemon::start_daemon::{{closure}}::{{closure}}::hf6ff73e9d938cf70
  14:     0x561753d85c6c - <std::future::GenFuture<T> as core::future::future::Future>::poll::h3a90b359628c0665
  15:     0x561753dfe2bd - tokio::runtime::handle::Handle::enter::h73797630f178e01b
  16:     0x561753e07c2a - rog_core::main::h0bc96bfba6b15cd4
  17:     0x561753df8936 - std::rt::lang_start::{{closure}}::hb66d0190f5773612
  18:     0x561753e07867 - main
  19:     0x7f35a37be002 - __libc_start_main
  20:     0x561753d7509e - _start
  21:                0x0 - <unknown>
fish: 'sudo env RUST_BACKTRACE=full /u…' zakończony sygnałem SIGABRT (Przerwij)
```

flukejones, 2020-05-22:
Was it working previously? Did you update then it stopped working? Kernel change?

dragonnn, 2020-05-22:
It was working fine, no updates where made and no kernel changes. They only thing with I was doing is working on a program to give some overlay on desktop over fan modes and others. It is just reading sysfs files, but I tried disabling it and no change.

flukejones, 2020-05-22:
`Invalid parameter` is a USB error code, but I'm unsure where exactly it is coming from. There's a handful of locations where it might be, but I'll need to add some more verbose messages.

Starting to wish I'd gone a bit harder on the error messages now.

flukejones, 2020-05-22:
If you're okay with building from source (master), do `sudo cargo run -- -d`

Also, what distro, kernel version etc?

dragonnn, 2020-05-23:
Arch, kernel 5.6.13. What is super strange, boot my laptop this morning and now it is back working.
I will report when it will crash again with the latest error.
You say USB error code, maybe some USB devices did go into a "strange state" and just stooped working right? And reboot didn't reset it but a full power down for a longer time did.

flukejones, 2020-05-23:
Really not sure what to make of this. Sounds like a possible malfunction somewhere else... If it happens again, please re-open the issue.

dragonnn, 2020-05-24:
Just got it on a new version,
```
[INFO ] Found GA401 series
[INFO ] Config loaded
[INFO ] INTERVAL: 1
[INFO ] MAX_PKT_SIZE: 64
[INFO ] SYNC: NoSync
[INFO ] TRANSFER_TYPE: Interrupt
[INFO ] ENDPOINT: 4
[ERROR] Could not claim device interface: InvalidParam
[ERROR] Invalid parameter
thread 'main' panicked at 'Invalid parameter', <::std::macros::panic macros>:5:6
stack backtrace:
   0:     0x56142b13be3a - backtrace::backtrace::libunwind::trace::h84fcc841152fc264
                               at /build/rust/src/rustc-1.43.1-src/vendor/backtrace/src/backtrace/libunwind.rs:86
   1:     0x56142b13be3a - backtrace::backtrace::trace_unsynchronized::h9512da4a7923cf5a
                               at /build/rust/src/rustc-1.43.1-src/vendor/backtrace/src/backtrace/mod.rs:66
   2:     0x56142b13be3a - std::sys_common::backtrace::_print_fmt::h80a9838fbf6210c7
                               at src/libstd/sys_common/backtrace.rs:78
   3:     0x56142b13be3a - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hfbfe9ad65a0640ca
                               at src/libstd/sys_common/backtrace.rs:59
   4:     0x56142b0b24dc - core::fmt::write::hb409a30975c0f9f5
                               at src/libcore/fmt/mod.rs:1063
   5:     0x56142b13b2a4 - std::io::Write::write_fmt::h7ee7c68a0296cf9a
                               at src/libstd/io/mod.rs:1426
   6:     0x56142b13aeaa - std::sys_common::backtrace::_print::h7cd9cb89129f3e0f
                               at src/libstd/sys_common/backtrace.rs:62
   7:     0x56142b13aeaa - std::sys_common::backtrace::print::ha9beca6e65f7f1cd
                               at src/libstd/sys_common/backtrace.rs:49
   8:     0x56142b13aeaa - std::panicking::default_hook::{{closure}}::h5e5a680f03f4adf9
                               at src/libstd/panicking.rs:204
   9:     0x56142b13a5c3 - std::panicking::default_hook::hb8f06365ed415b3d
                               at src/libstd/panicking.rs:224
  10:     0x56142b13a5c3 - std::panicking::rust_panic_with_hook::h1e03621439587007
                               at src/libstd/panicking.rs:470
  11:     0x56142b13a1aa - rust_begin_unwind
                               at src/libstd/panicking.rs:378
  12:     0x56142b13a150 - std::panicking::begin_panic_fmt::hc43e1369214a6caa
                               at src/libstd/panicking.rs:332
  13:     0x56142b095a11 - daemon::daemon::start_daemon::{{closure}}::{{closure}}::h0a0408030f3a9854
  14:     0x56142b0a584a - rog_core::main::{{closure}}::hb1beb99279287350
  15:     0x56142b125e36 - tokio::runtime::context::enter::h0505328d0ffcff1a
  16:     0x56142b124213 - rog_core::main::h8f90d077d4d63f72
  17:     0x56142b11af06 - std::rt::lang_start::{{closure}}::h7668ee5887648bd6
  18:     0x56142b123d37 - main
  19:     0x7f07ce894002 - __libc_start_main
  20:     0x56142b09209e - _start
  21:                0x0 - <unknown>
fish: 'sudo env RUST_BACKTRACE=full /u…' zakończony sygnałem SIGABRT (Przerwij)
```
I think this happens at this situation: using laptop on battery, suspend it, going to connect charger and wakening up my laptop (but after I connect the charger). Then I am doing a system reboot and rog-core doesn't start any more. 
To fix it I make a shutdown and it is working again. This sound like a hardware that goes into wrong state but maybe we can somehow work around it.

dragonnn, 2020-05-26:
I don't see how to reopen the issue, but I found a workaround without need to shutdown the system. Running those commands brings rog-core back:
```
echo -n "0000:04:00.3" | sudo tee /sys/bus/pci/drivers/xhci_hcd/unbind
sleep 5
echo -n "0000:04:00.3" | sudo tee /sys/bus/pci/drivers/xhci_hcd/bind
```
I have found those because somehow when I used my stlink v3 programmer with openocd and gdb it make the USB Host die :/. dmesg reported those:
```
[ 5010.211173] xhci_hcd 0000:04:00.3: xHCI host not responding to stop endpoint command.
[ 5010.214494] xhci_hcd 0000:04:00.3: xHCI host controller not responding, assume dead
```
And all ASUS devices on USB have vanished and everything on on the two USB ports on the right. Keyboard also stooped working, only touchpad worked and everything connect to the USB-C port on the left.

flukejones, 2020-05-28:
I wonder if getting `rog-core` to hand the device back to the kernel before sleep helps.

When you get a chance, try `systemctl stop rog-core.service` before sleep, and see if that helps. In theory libusb (which rog-core uses) **should** hand the device back to the kernel.

If that doesn't work I might need to add some way to detect when the system is going in to sleep mode and release the device before doing so.

Does updating kernel have any effect? I use Ubuntu only, and kernel 5.6.14 at the moment.

dragonnn, 2020-05-28:
I have 5.6.14 kernel too. 
For now I just run: systemctl edit rog-core and add this text:
```
[Service]                                                                                                                                                    
ExecStartPre=-/usr/local/bin/xhci_restart
```
And the xhci_restart file:
```
#!/bin/bash
echo -n "0000:04:00.3" | sudo tee /sys/bus/pci/drivers/xhci_hcd/unbind
sleep 0.5
echo -n "0000:04:00.3" | sudo tee /sys/bus/pci/drivers/xhci_hcd/bind
sleep 3.5
```
Note - that problem appers not directly after sleep, but after a reboot after sleep. But for me this looks like it fixed. Does systemd have some sort ExecCrash? 
I will try if a hook before sleep disabling rog-core will help

flukejones, 2020-06-25:
@dragonnn is this still happening?

AstroProfundis, 2020-06-28:
It seems whit the latest v0.12.0 the USB devices are reset randomly when the device is not suspending or hibernating, and that makes my bluetooth keyboard and wireless (via 2.4G) mouse not working for some seconds.

I can see in dmesg the USB devices are reset, but no logs in rog-core, so I can not confirm if it was caused by rog-core, but it seems not happening before I upgrade (and not when I stop rog-core).

flukejones, 2020-07-05:
@AstroProfundis, I'm not really able to find a way to replicate this, so is it still happening in the most recent release (0.14.3)?

AstroProfundis, 2020-08-10:
It's been some time, sorry for that as I've forgot this issue. I'm now using 0.15.3 and seems not having the problem for quite some time, I'd say it has been fixed on my machine. Though I don't know exactly when did it got fixed.

closed 2020-08-10 by flukejones