flukejones/rog-core #9: Setting turbo/boost on AMD gives an error

dragonnn, 2020-05-23:
```
[ERROR] Could not write to /sys/devices/system/cpu/cpufreq/boost, Os { code: 22, kind: InvalidInput, message: "Invalid argument" }
```
I think it means it is writing wrong value?
BTW. it would be fine to also allow setting no_turbo to for example "ignore" or "donotchange" or just comment it out.

flukejones, 2020-05-23:
From what I was reading about it, you can just write a 0 or 1 to it? It might be a case of it expecting a 1 as a char... I'll do a quick fix for you to check.

dragonnn, 2020-05-23:
0 or 1 but as char

flukejones, 2020-05-23:
Got it. Will fix shortly, easy fix.

flukejones, 2020-05-23:
[Pre-release here](https://github.com/flukejones/rog-core/releases/tag/v0.9.9)

dragonnn, 2020-05-24:
Works now, thank you. Did you think about the "ignore" option? I wrote my own program for creating power management profiles: https://gitlab.com/dragonn/rog-watcher it is still pretty basic but works like I like it.

closed 2020-05-24 by dragonnn