flukejones/rog-core #10: G401IV - aniMe Matrix Display

aspann, 2020-05-24:
I finally got the windows software (Armoury Crate) and services running. (it was a pain)

Howto install "Armory Crate" on a "fresh" windows installation. (not the preinstalled one):

1. Turn on Secureboot (BIOS)
2. Turn on "Armoury Crate interface support" (BIOS)
3. Clean windows installation
4. Install all upates and drivers (amd/Nvidia) 
5. enable .net 2.0/3.0 windows feature
6. Install "ROG Font" and "ASUSSystemControlInterfaceV2_DCH_W10_64_V22100"
7. Reboot
8. install "ROGLiveServicePackage_W10_64_V10250"
9. Reboot
10. Install "ARMOURYCRATEService_ASUS_V272"
11. Reboot
12. install "Armoury Crate" UWP 
13. Reboot
14. Waiit upto 30 Minutes to allow "Armoury Crate" fetching its updates 
15. ready to use..

I'll post any progress using this issue. 

[quote]:
> @aspann for the animatrix thingy I'm expecting something like [this](https://github.com/flukejones/rog-core/blob/master/wireshark_data/gx501/per_key_raw_bytes). But since it's a single colour it should be a lot simpler... maybe. I've no idea of the resolution of it. Maybe if we can get the corner LEDs byte sequence it'll give an idea of a pattern.
> 
> From memory, the packet sent to USB is 64 bytes, and at least 57 of those bytes are used for LED control (on the GX502). If the display is written by bit pattern, then it can control 456 of those pixels per packet - which means it's likely to be multiple packets. The GX502 uses ten packets to control the RGB...
> 
> There's likely to be an initialization byte sequence, and I suspect it will be part of the keyboard controller, which may make things a little easier.

aspann, 2020-05-31:
The aniMe Matrix(tm) Display is working and obviously fully functional.

PoC: https://lab.retarded.farm/zappel/asus-rog-zephyrus-g14/-/blob/master/anime_matrix_display/README.md

cheers.

tuxuser, 2020-05-31:
Visual PoC (GIF)

[image]

flukejones, 2020-05-31:
Work is in progress to integrate what we have so far [here](https://github.com/flukejones/rog-core/tree/animatrix).  I can only do so much though, but the important part of:

- add to daemon
- set up G14 laptop data
- methods for init and write
- dbus commands (server)
- dbus commands (client)

are done. Now what is required is methods to convert images to required byte format.

flukejones, 2020-05-31:
Also need to implement:

- save last image write (if this is a desired thing by anyone, so that it loads on boot)
- Implement SET and APPLY for dbus commands

I'm unsure if a SET or APPLY need to run after writing the two arrays. Someone will need to clarify this for me.

aspann, 2020-06-01:
Frame timing measurement (as requested):
>> avg: 21.369288ms, min: 18.884827ms, max: 23.153204ms

Measuerd over a cycle of 255 Frames, each "frame" consists out of two panes and one flush([0x5e, 0xc0, 0x03]).

I'm testing the DBus thingy as well, results will follow.

aspann, 2020-06-01:
Thanks to @tuxuser, he adopted the PoC to dbus:

```rust
use std::error::Error;
use std::time::{Duration, Instant};
use rog_aura::AnimatrixDbusWriter;

fn write_via_dbus(
    pixel_data: &[u8]
) -> Result<Duration, Box<dyn Error>> {
    if pixel_data.len() != 1266 {
        println!("doh!");
    }
    let prev_time = Instant::now();

    // Assemble the two packets
    let pane1_prefix: &[u8] = &[0x5e,0xc0,0x02,0x01,0x00,0x73,0x02];
    let pane2_prefix: &[u8] = &[0x5e,0xc0,0x02,0x74,0x02,0x73,0x02];

    let pane1_data: &mut [u8; 640] = &mut [0; 640];
    let pane2_data: &mut [u8; 640] = &mut [0; 640];

    // Copy pane1 prefix & data
    pane1_data[..7].copy_from_slice(pane1_prefix);
    pane1_data[7..640].copy_from_slice(&pixel_data[..633]);

    // Copy pane2 prefix & data
    pane2_data[..7].copy_from_slice(pane2_prefix);
    pane2_data[7..640].copy_from_slice(&pixel_data[633..]);

    /* rog-core example code */
    let mut writer = AnimatrixDbusWriter::new()?;

    let mut send_buffer = [vec![0; 0], vec![0; 0]];
    send_buffer[0].extend(pane1_data.iter().cloned());
    send_buffer[1].extend(pane2_data.iter().cloned());

    writer.write_image(&send_buffer)?;
    return Ok(Instant::now().duration_since(prev_time));
}

fn main() {
    loop {
        for brightness in 0..0xFF {
            let buf: &[u8; 1266] = &[brightness; 1266];
            match write_via_dbus(buf) {
                Ok(s) => println!("{:?}", s),
                Err(e) => println!(">> Error: {}", e.to_string()),
            }
        }
    }
}
```

Avg. frame time: 103 ms

cheers.

flukejones, 2020-06-01:
I only required timing for the actual write to USB.

Just a tip though, the code above does *a lot* of memory allocs. It can be very concisely written as:

```rust
use rog_aura::AnimatrixDbusWriter;

const PANE1_PREFIX: [u8; 7] = [0x5e, 0xc0, 0x02, 0x01, 0x00, 0x73, 0x02];
const PANE2_PREFIX: [u8; 7] = [0x5e, 0xc0, 0x02, 0x74, 0x02, 0x73, 0x02];

fn main() {
    // pass a ref to this rather than recreate each time
    let mut writer = AnimatrixDbusWriter::new().unwrap();

    loop {
        for brightness in 0..0xFF {
            let mut buffers = [vec![brightness; 640], vec![brightness; 640]];
            buffers[0][..7].copy_from_slice(&PANE1_PREFIX);
            buffers[0][..7].copy_from_slice(&PANE2_PREFIX);

            writer.write_image(&buffers).unwrap();
        }
    }
}
```

flukejones, 2020-06-01:
I will adjust dbus blocking time to 25ms, this will give a 40fps without any missed writes hopefully.

flukejones, 2020-06-09:
Closing issue as solved from v0.10.0. All confirmed working.

closed 2020-06-09 by flukejones