flukejones/rog-core #12: Zephyrus G15 (Ga502)

SinaQane, 2020-06-02:
G15 has problems dealing with Ubuntu as well:

1. Overheating due to inactivity of fans in Performance mode or Turbo mode
2. All of fn keys and RoG key are not supported (You can only use volume keys)
3. It works unbelievably slow (for example it takes 5-6 seconds to just open Chromium)

I think the Wifi and GPU and CPU drivers don't get installed on it by default. (Mine had a really major problem not recognizing Wifi and I couldn't even open Settings (because it would freeze since there was no GPU driver) and I had to solve it by connecting LAN and turn off/on for a hundred times to install Wifi and LAN drivers).

flukejones, 2020-06-02:
It sounds like a lot of issues will come from the kernel itself, especially if it's fan or CPU related. I strongly advise you to try the 5.7 series kernel, and you can use [ukuu](https://teejeetech.in/ukuu/) to help there.

This is a fairly new laptop with a new CPU type, and other new chipsets. It might take some time for the required code to make it in to the kernels used. Please keep in mind that `rog-core` focuses mostly on the  fn keys, RGB LEDs, and AniMe matrix.

It's highly likely that rog-core will work for most things. But to be sure please provide details from the following:

First do `lsusb |grep 0b05` and check the part after `0b05:`, output looks like:

```text
Bus 001 Device 005: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
```

Then do `sudo lsusb -vd 0b05:1866 > ~/laptop_info` and give that to me. If there is more than one device for `lsusb |grep 0b05` please provide both.

Other helpful info can be gained from `sudo usbhid-dump`, for which you may need to unload kernel drivers. Please google this.

Also required (for my book-keeping of data):
- `cat /sys/class/dmi/id/product_name`
- `cat /sys/class/dmi/id/product_family`
- `cat /sys/class/dmi/id/board_name`

dragonnn, 2020-06-02:
Ubuntu is a poor choice for such new laptop.
I sugest somethinkg like Manjaro or Arch. And you still will need some kernel patches, checkout the G14 thread. They are posted patches need to get fan control and other things working. 
Sure they will be need to add G15 ids, but this can be easly done if you provide the right information.

flukejones, 2020-06-02:
One could argue that Manjaro or Arch are poor choices for new users also :wink: .

It looks like the G15 is just a 15" version of the G14?

aspann, 2020-06-02:
This issue might be related to a non-working asus(-nb)-wmi module in combination with a wrong thermalzone setting. 

Could you please post the output of `dmseg | grep wmi` as well as the output of `/sys/devices/platform/asus-nb-wmi/throttle_thermal_policy`. If the sys-device is available a quick 'n dirty (non-persistnent) fix might be: `echo 1 > /sys/devices/platform/asus-nb-wmi/throttle_thermal_policy` if asus-nb-wmi is working. 

cheers.

PS:
@dragonnn you forgot about gentoo! :rofl:

isushik94, 2020-06-07:
Hi there.
I tried this utility on my G15 GA502IU laptop and everything worked fine except 1) fan mode toggling and 2) sometimes something crashed during the service startup and the keyboard doesn't work at all. 
The first one didn't work due to a lack of `asus-nb-wmi`. As I found it depends on `i8042` which isn't loaded on my laptop (I'll try the patch from https://github.com/flukejones/rog-core/issues/3#issuecomment-631699265).
The second issue I'll check again after fixing the first one. I only can say that it was a rust error according to the utility's log entries.

Here is some info requested:

`cat /sys/class/dmi/id/product_name`
```
ROG Zephyrus G15 GA502IU_GA502IU
```

`cat /sys/class/dmi/id/product_family`
```
ROG Zephyrus G15
```

`cat /sys/class/dmi/id/board_name`
```
GA502IU
```

[dmidecode.txt](https://github.com/flukejones/rog-core/files/4742307/dmidecode.txt)

isushik94, 2020-06-07:
Here is a backtrace of the crash I've mentioned before:

```
Jun 07 22:37:33 asus-rog rog-core[770]: [INFO ] Config loaded
Jun 07 22:37:33 asus-rog rog-core[770]: [INFO ] INTERVAL: 1
Jun 07 22:37:33 asus-rog rog-core[770]: [INFO ] MAX_PKT_SIZE: 64
Jun 07 22:37:33 asus-rog rog-core[770]: [INFO ] SYNC: NoSync
Jun 07 22:37:33 asus-rog rog-core[770]: [INFO ] TRANSFER_TYPE: Interrupt
Jun 07 22:37:33 asus-rog rog-core[770]: [INFO ] ENDPOINT: 4
Jun 07 22:37:33 asus-rog rog-core[770]: [ERROR] Could not claim device interface: InvalidParam
Jun 07 22:37:33 asus-rog rog-core[770]: [ERROR] Invalid parameter
Jun 07 22:37:33 asus-rog rog-core[770]: thread 'main' panicked at 'Invalid parameter', /build/rust/src/rustc-1.44.0-src/src/libstd/macros.rs:16:9
Jun 07 22:37:33 asus-rog rog-core[770]: stack backtrace:
Jun 07 22:37:33 asus-rog rog-core[770]:    0:     0x559aa5733e3a - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    1:     0x559aa56adfac - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    2:     0x559aa5733334 - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    3:     0x559aa5732d6a - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    4:     0x559aa573250a - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    5:     0x559aa57320ea - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    6:     0x559aa5732090 - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    7:     0x559aa568f4f1 - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    8:     0x559aa56a1629 - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:    9:     0x559aa57177ed - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:   10:     0x559aa5742c56 - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:   11:     0x559aa5717175 - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:   12:     0x7f8c2d90a002 - __libc_start_main
Jun 07 22:37:33 asus-rog rog-core[770]:   13:     0x559aa568e0ae - <unknown>
Jun 07 22:37:33 asus-rog rog-core[770]:   14:                0x0 - <unknown>
```

isushik94, 2020-06-07:
This helped to overcome the crash issue: https://github.com/flukejones/rog-core/issues/8#issuecomment-635515744
But I had to change that bus number to `0000:06:00.3`

flukejones, 2020-06-08:
Thanks for the feedback!

`[ERROR] Could not claim device interface: InvalidParam` indicates that it is still an kernel issue with hardware drivers etc.

If you're able to, try using the `animatrix` branch, or at least v10+. There are some extra fixes in there along with the AniMe dot-matrix stuff.

I'm in the middle of a house move so the next two weeks i'll be a bit quiet.

dragonnn, 2020-06-08:
> This helped to overcome the crash issue: [#8 (comment)](https://github.com/flukejones/rog-core/issues/8#issuecomment-635515744)
> But I had to change that bus number to `0000:06:00.3`

For G14 on the ASUS page there is an update for the keyboard firmware. I didn't check if it helps, need to install Windows back. Maybe they is such update for G15

@flukejones do you trigger usb reset when claiming the device. I mean something like this example https://github.com/ralight/usb-reset
Maybe this would help with that issue.

isushik94, 2020-06-08:
> For G14 on the ASUS page there is an update for the keyboard firmware. I didn't check if it helps, need to install Windows back. Maybe they is such update for G15

I've just checked and there is only bios update. But after adding of the usb resetting script everything works fine.

flukejones, 2020-06-09:
@isushik94 can you give a bit more detail on how you solved the issue with usb reset? When do you need to do it? Is it before running rog-core, or after sleep/wake etc?

I've added a force reset just after the device grab, so let's see if that helps. This is in the upcoming v0.11.0 release tonight.

isushik94, 2020-06-09:
Hi, @flukejones.

I've added a prestart script as in this [comment](https://github.com/flukejones/rog-core/issues/8#issuecomment-635515744). So, it runs before starting the rog-core's systemd unit during linux initialization. But I haven't checked how it works after hibernation.

I'll check v0.11.0 and report you.

isushik94, 2020-06-09:
I've checked and found something interesting

I've got a release build of the utility and after reboot it's produced the same error I had before:

```
Jun 09 15:41:29 asus-rog rog-core[802]: [INFO ] Found GX502 or GA502 series
Jun 09 15:41:29 asus-rog rog-core[802]: [INFO ] Config loaded
Jun 09 15:41:29 asus-rog rog-core[802]: [INFO ] INTERVAL: 1
Jun 09 15:41:29 asus-rog rog-core[802]: [INFO ] MAX_PKT_SIZE: 64
Jun 09 15:41:29 asus-rog rog-core[802]: [INFO ] SYNC: NoSync
Jun 09 15:41:29 asus-rog rog-core[802]: [INFO ] TRANSFER_TYPE: Interrupt
Jun 09 15:41:29 asus-rog rog-core[802]: [INFO ] ENDPOINT: 4
Jun 09 15:41:29 asus-rog rog-core[802]: [ERROR] Could not claim keyboard device interface: InvalidParam
Jun 09 15:41:29 asus-rog rog-core[802]: [ERROR] Invalid parameter
Jun 09 15:41:29 asus-rog rog-core[802]: thread 'main' panicked at 'Invalid parameter', /build/rust/src/rustc-1.44.0-src/src/libstd/macros.rs:16:9
Jun 09 15:41:29 asus-rog rog-core[802]: note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Jun 09 15:41:30 asus-rog systemd[1]: rog-core.service: Main process exited, code=dumped, status=6/ABRT
```

`RUST_BACKTRACE=full` hasn't added something meaningful, so I've decided to build a debug version for a better stacktrace: `DEBUG=1 make`. And after reboot the keyboard's worked. So I've rebooted my laptop a few times and this behavior repeated.

flukejones, 2020-06-09:
I wonder if it's a timing issue.

Can you try a new release build and in `rog-core/src/rogcore.rs`, above line [72](https://github.com/flukejones/rog-core/blob/d53cef82685822b09fbf06fec346cbe74a2af5b5/rog-core/src/rogcore.rs#L72) put:

```
std::thread::sleep(std::time::Duration::from_secs(1)); // maybe two seconds?
```

Line [41](https://github.com/flukejones/rog-core/blob/d53cef82685822b09fbf06fec346cbe74a2af5b5/rog-core/src/rogcore.rs#L41) might need to be moved to below `pub fn new`...

flukejones, 2020-06-09:
It also might be a systemd ordering issue..? Maybe we can discover at which precise point rog-core can run.

isushik94, 2020-06-11:
The solution I found, which works for many reboots, is to use the udev rule:
```
ACTION=="bind", SUBSYSTEM=="usb", ATTRS{authorized}=="1",  ATTR{idVendor}=="0b05", ENV{ROG_USB_KEYBOARD}="1"

ENV{ROG_USB_KEYBOARD}=="1", TAG+="systemd", ENV{SYSTEMD_WANTS}="rog-core.service"
```
And here is the output of `udevadm info --attribute-walk --name=/dev/input/by-id/usb-ASUSTeK_Computer_Inc._N-KEY_Device-event-kbd` for the keyboard device: 
[udevadm-info.txt](https://github.com/flukejones/rog-core/files/4764046/udevadm-info.txt)

isushik94, 2020-07-05:
After doing some investigation I've probably found something. I inserted the code right before `dev_handle.set_auto_detach_kernel_driver(true)`:

```rust
if let Ok(active) = dev_handle.kernel_driver_active(interface) {
    if !active {
        dev_handle.attach_kernel_driver(interface).map_err(|err| {
            warn!(
                "Could not attach kernel driver for RAII detachment: {:?}",
                err
            );
            err
        })?;
    }
}
```

And I've got the following log during system startup when the keyboard is freezed:

```
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: Product name: ROG Zephyrus G15 GA502IU_GA502IU
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: Board name: GA502IU
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: No RGB control available
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: Config loaded
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004436] [000002b8] libusb: debug [libusb_get_device_list]
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004443] [000002b8] libusb: debug [libusb_get_device_descriptor]
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004447] [000002b8] libusb: debug [libusb_get_device_descriptor]
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004452] [000002b8] libusb: debug [libusb_get_device_descriptor]
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004456] [000002b8] libusb: debug [libusb_get_device_descriptor]
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004460] [000002b8] libusb: debug [libusb_get_device_descriptor]
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004465] [000002b8] libusb: debug [libusb_open] open 3.3
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004484] [000002b8] libusb: debug [usbi_add_pollfd] add fd 12 events 4
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.004490] [000002b8] libusb: debug [libusb_set_configuration] configuration 0
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.005473] [000002b8] libusb: debug [libusb_get_config_descriptor] index 0
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: C-struct:
Jul 05 14:55:27 asus-rog rog-core[696]: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 131, bmAttributes: 3, wMaxPacketSize: 64, bInterval: 1 }
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: INTERVAL: 1
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: MAX_PKT_SIZE: 64
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: SYNC: NoSync
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: TRANSFER_TYPE: Interrupt
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: ENDPOINT: 83
Jul 05 14:55:27 asus-rog rog-core[696]: INFO: INTERFACE: 2
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.005533] [000002b8] libusb: debug [libusb_kernel_driver_active] interface 2
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.005542] [000002b8] libusb: debug [libusb_attach_kernel_driver] interface 2
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.005548] [000002b8] libusb: error [op_attach_kernel_driver] attach failed error -1 errno 113
Jul 05 14:55:27 asus-rog rog-core[696]: WARN: Could not attach kernel driver for RAII detachment: Other
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.005558] [000002b8] libusb: debug [libusb_close]
Jul 05 14:55:27 asus-rog rog-core[696]: [ 0.005564] [000002b8] libusb: debug [usbi_remove_pollfd] remove fd 12
Jul 05 14:55:27 asus-rog rog-core[696]: ERROR: Other error
```

But if I start the daemon after I logged into my DE I got the next:

```
Jul 05 14:57:58 asus-rog rog-core[1996]: INFO: C-struct:
Jul 05 14:57:58 asus-rog rog-core[1996]: EndpointDescriptor { bLength: 7, bDescriptorType: 5, bEndpointAddress: 131, bmAttributes: 3, wMaxPacketSize: 64, bInterval: 1 }
Jul 05 14:57:58 asus-rog rog-core[1996]: INFO: INTERVAL: 1
Jul 05 14:57:58 asus-rog rog-core[1996]: INFO: MAX_PKT_SIZE: 64
Jul 05 14:57:58 asus-rog rog-core[1996]: INFO: SYNC: NoSync
Jul 05 14:57:58 asus-rog rog-core[1996]: INFO: TRANSFER_TYPE: Interrupt
Jul 05 14:57:58 asus-rog rog-core[1996]: INFO: ENDPOINT: 83
Jul 05 14:57:58 asus-rog rog-core[1996]: INFO: INTERFACE: 2
Jul 05 14:57:58 asus-rog rog-core[1996]: [ 0.002608] [000007cc] libusb: debug [libusb_kernel_driver_active] interface 2
Jul 05 14:57:58 asus-rog rog-core[1996]: [ 0.002616] [000007cc] libusb: debug [libusb_claim_interface] interface 2
Jul 05 14:57:59 asus-rog rog-core[1996]: INFO: RogCore loaded
```

So during start up the kernel driver is not attached and probably is not properly initialized yet to allow usage of the usb device. I haven't found a possible fix. Waiting (with sleep for 100ms) in a loop doesn't work. The service is shut down by systemd or something else after a while.

flukejones, 2020-07-05:
@isushik94 What distro are you using?

flukejones, 2020-07-05:
Can someone try this:

Create file `/etc/udev/rules.d/99-rogcore.rules` with:

```
ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0b05", ATTRS{idProduct}=="1866", TAG+="systemd", ENV{SYSTEMD_WANTS}="rog-core.service"
```

and disable the rog-core service. I've only tested this on Ubuntu, works there, but the thing I'm wondering is where in the systemd init chain this will actually run.

I did `sudo systemd-analyze plot > ordering.svg` to see where things were ending up, and for me rog-core is in the final block (right down the bottom) with this udev rule, and seems to be initialising a little sooner than with just the unit being enabled.

Gades0, 2020-07-07:
Hi @flukejones  
I followed your instructions and here is a the ordering svg from my system, I have an G15 (model GA502IV)

[image]

flukejones, 2020-07-07:
> Hi @flukejones
> I followed your instructions and here is a the ordering svg from my system, I have an G15 (model GA502IV)

Thanks, were you having issues before this?

flukejones, 2020-07-07:
@isushik94 can you have a try with standard systemd service enabled using the latest master please? I'm curious to know at what point the fail occurs.

Gades0, 2020-07-07:
No I had no issue, but I had the udev rule propose by isushik94 configured and the rog-core service enable.

flukejones, 2020-07-07:
> No I had no issue, but I had the udev rule propose by isushik94 configured and the rog-core service enable.

I guess this might be what is required for some configurations in the end. Would you be so kind as to try without the udev rule? Make sure you have a usb keyboard handy in case it kills input.

Gades0, 2020-07-07:
I have remove the udev rules, enable the rog-core service, and everything boot correct.
Do you want another ordering plot file?
I think my previous issue was that I had to an old rog-core configuration and it made it crash. So the udev rules might not have fix anything for me

flukejones, 2020-07-07:
Hmm, if you could do both the plot and `sudo journalctl -b -u rog-core > rog.log` (and attach both) I would super appreciate it.

Gades0, 2020-07-07:
Here are the files you requested
[rog.log](https://github.com/flukejones/rog-core/files/4882243/rog.log)

[image]

flukejones, 2020-07-07:
@Gades0, what distro?

Gades0, 2020-07-07:
Archlinux with dragonn G14 patched kernel

flukejones, 2020-07-07:
Great, thanks. Would you mind leaving the current service active instead of the udev rule for a wee bit, and try things like sleep/wake/reboot? After a wake/reboot a quick `sudo journalctl -b -u rog-core |grep warn` will tell you if I need a new `sudo journalctl -b -u rog-core > rog.log`.

If you're not there already, the discord channel is [here](https://discord.gg/zC7K4e)

isushik94, 2020-07-07:
@flukejones, sorry for the late response. I was super busy.

> What distro are you using?
It's arch linux. I've been at the discord channel and I've promised to investigate this issue.

> Create file /etc/udev/rules.d/99-rogcore.rules with:
I did it before as a workaround but "add" action sometimes stopped working and then I found that there is "bind" action and it works so far without problem. Btw my rule named as 90-xxxx.rules

> @isushik94 can you have a try with standard systemd service enabled using the latest master please?
Yes, I'll try this evening (in 6 hours)

isushik94, 2020-07-07:
@Gades0, could you explain how you get his plot?

flukejones, 2020-07-07:
> @Gades0, could you explain how you get his plot?

Systemd command [here](https://github.com/flukejones/rog-core/issues/12#issuecomment-653946255)

flukejones, 2020-07-07:
> I did it before as a workaround but "add" action sometimes stopped working and then I found that there is "bind" action and it works so far without problem.

Hmm, i wasn't able to get it working on Ubuntu with the "bind". Given how USB behaves I guess that makes sense, I suspect Arch does other things in init that Ubuntu and others do not. Maybe bind forces a slightly longer wait for full device startup to complete.

isushik94, 2020-07-07:
> Systemd command here

Thanks

> Hmm, i wasn't able to get it working on Ubuntu with the "bind".

I've found [this](https://stackoverflow.com/questions/3389192/register-bind-match-a-device-with-a-driver) and it made me thinking that we can check whether a kernel driver is loaded with `DRIVER=="xxx"` but I have no idea what kernel driver we need.

isushik94, 2020-07-07:
@flukejones, I composed another udev rule which works (at least a few times):

`ACTION=="add|change", SUBSYSTEM=="input", ENV{ID_VENDOR_ID}=="0b05", ENV{ID_MODEL_ID}=="18[0-9][0-9]", ENV{ID_TYPE}=="hid", TAG+="systemd", ENV{SYSTEMD_WANTS}="rog-core.service"`

Try this on Ubuntu.

P.S. Action `change` is for manual reloading with `udevadm trigger`.

isushik94, 2020-07-07:
I've also added [this](https://blog.dachary.org/2016/11/30/logging-udev-events-at-boot-time/) service to record udev log during initialization.

[udev_monitor.log](https://github.com/flukejones/rog-core/files/4886773/udev_monitor.log)

flukejones, 2020-07-08:
> I've also added [this](https://blog.dachary.org/2016/11/30/logging-udev-events-at-boot-time/) service to record udev log during initialization.
> 
> [udev_monitor.log](https://github.com/flukejones/rog-core/files/4886773/udev_monitor.log)

Top stuff. Thanks for looking that up. So, I get a very different init to you - there are no bind events at all (except for unbind).

Subsystem input is later init than usb?

[udev_monitor.log](https://github.com/flukejones/rog-core/files/4889688/udev_monitor.log)

flukejones, 2020-07-08:
I'm assuming you suggested the new rule since it works for you. Confirmed working on Ubuntu fine too so I'll include it and make this the default init method.

isushik94, 2020-07-08:
@flukejones, yes, it works for me.

flukejones, 2020-07-10:
Great! I think we can close this now.

closed 2020-07-10 by flukejones