flukejones/rog-core #14: 0.10.0 says G401

gree7, 2020-06-07:
```
root@zephy /h/g/d/rog-core-0.9.9 [2]# rog-core -v
Version: 0.10.0
root@zephy /h/gree [1]# rog-core -d
[INFO ] Found GA401 series
[INFO ] Config loaded
[INFO ] INTERVAL: 1
[INFO ] MAX_PKT_SIZE: 64
[INFO ] SYNC: NoSync
[INFO ] TRANSFER_TYPE: Interrupt
[INFO ] ENDPOINT: 4
[INFO ] RogCore loaded
[WARN ] Fan mode: Fan mode not available
[INFO ] Reloaded last used mode and brightness
[ERROR] Could not get device handle: NoDevice
Error: NoDevice
```

gree7, 2020-06-07:
I am available to help with fixing the problem. I have also some rudimentary rust knowledge, if needed. Just let me know.

flukejones, 2020-06-08:
Hmm, looks like a result of me thinking that all GA401 devices had the AniMe matrix display. I'll have a think about how better to handle this, but for now can you do:

```
    cat /sys/class/dmi/id/product_name
    cat /sys/class/dmi/id/product_family
    cat /sys/class/dmi/id/board_name
```

The root cause is that the USB device `0x0B05:0x193b` is assumes to be present, which it isn't here. To get you going, if you can build from source, comment out this [line](https://github.com/flukejones/rog-core/blob/ffb62bbcdd449b3dfdfbdf8ab0dd0df815e11529/rog-core/src/laptops.rs#L88)

dragonnn, 2020-06-08:
Well no, only the top end G14 have animatrix, mine don't have it too but I didn't update rog-core yet.
```
ROG Zephyrus G14 GA401IU_GA401IU
ROG Zephyrus G14
GA401IU
```

flukejones, 2020-06-08:
Looks like the `GA401` with `I<letter>` prefix might be no AniMe? Need to confirm with @aspann what their model is. I Thought I had it but can't find it in my notes.

gree7, 2020-06-08:
It does not seems so... `GA401IV-AniMe177T ` is with AniMe and `GA401IV-HE003T ` is without.

flukejones, 2020-06-08:
Thanks, okay I'll add a failsafe check to if the device exists instead.

gree7, 2020-06-08:
That sounds like a safer solution. Thanks a lot.

flukejones, 2020-06-08:
@gree7 if you're okay with building from source, please test the animatrix branch with fix.

gree7, 2020-06-08:
I'll do, no problem (I am on meeting though, give me some time).

gree7, 2020-06-08:
```
gree@zephy ~/d/r/t/release (animatrix) [SIGABRT]> RUST_BACKTRACE=1 rog-core -d
[INFO ] Found GA401 series
thread 'main' panicked at 'config file error: Os { code: 13, kind: PermissionDenied, message: "Permission denied" }', src/libcore/result.rs:1188:5
stack backtrace:
   0: backtrace::backtrace::libunwind::trace
             at /usr/src/rustc-1.41.0/vendor/backtrace/src/backtrace/libunwind.rs:88
   1: backtrace::backtrace::trace_unsynchronized
             at /usr/src/rustc-1.41.0/vendor/backtrace/src/backtrace/mod.rs:66
   2: std::sys_common::backtrace::_print_fmt
             at src/libstd/sys_common/backtrace.rs:84
   3: <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt
             at src/libstd/sys_common/backtrace.rs:61
   4: core::fmt::write
             at src/libcore/fmt/mod.rs:1025
   5: std::io::Write::write_fmt
             at src/libstd/io/mod.rs:1426
   6: std::sys_common::backtrace::_print
             at src/libstd/sys_common/backtrace.rs:65
   7: std::sys_common::backtrace::print
             at src/libstd/sys_common/backtrace.rs:50
   8: std::panicking::default_hook::{{closure}}
             at src/libstd/panicking.rs:193
   9: std::panicking::default_hook
             at src/libstd/panicking.rs:210
  10: std::panicking::rust_panic_with_hook
             at src/libstd/panicking.rs:471
  11: rust_begin_unwind
             at src/libstd/panicking.rs:375
  12: core::panicking::panic_fmt
             at src/libcore/panicking.rs:84
  13: core::result::unwrap_failed
             at src/libcore/result.rs:1188
  14: rog_core::main::{{closure}}
  15: rog_core::main
  16: std::rt::lang_start::{{closure}}
  17: main
  18: __libc_start_main
  19: _start
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
fish: “RUST_BACKTRACE=1 rog-core -d” terminated by signal SIGABRT (Abort)
gree@zephy ~/d/r/t/release (animatrix) [SIGABRT]> 
```

gree7, 2020-06-08:
This is, I suppose, unrelated:
```
[WARN ] Custom { kind: NotFound, error: "Fan mode not available" }
```

flukejones, 2020-06-08:
> This is, I suppose, unrelated:
> 
> ```
> [WARN ] Custom { kind: NotFound, error: "Fan mode not available" }
> ```

To a point, it's certainly not `rog-core` fault, but the info for that is in this [issue](https://github.com/flukejones/rog-core/issues/3) somewhere. You'll need a few patches and a more recent kernel.

The lovely folks in there will be more than willingly to help out.

flukejones, 2020-06-08:
Closing as solved by cb8032c. New release pending.

closed 2020-06-08 by flukejones