flukejones/rog-core #15: G14 - Error: NoDevice

shoenig, 2020-06-09:
```
$ uname -a
Linux zg14 5.6.16-300.fc32.x86_64 #1 SMP Thu Jun 4 18:08:38 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

After building and trying to start the systemd service, it fails to start with

```
Jun 07 20:28:58 zg14 systemd[1]: rog-core.service: Scheduled restart job, restart counter is at 4.
Jun 07 20:28:58 zg14 systemd[1]: Stopped ROG Core Daemon.
Jun 07 20:28:58 zg14 systemd[1]: Started ROG Core Daemon.
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] Found GA401 series
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] Config loaded
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] INTERVAL: 1
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] MAX_PKT_SIZE: 64
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] SYNC: NoSync
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] TRANSFER_TYPE: Interrupt
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] ENDPOINT: 4
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] RogCore loaded
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] Setting pstate for AMD CPU
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] CPU Power: turbo: "1"
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] Reloaded fan mode: Normal
Jun 07 20:28:58 zg14 rog-core[1430]: [INFO ] Reloaded last used mode and brightness
Jun 07 20:28:58 zg14 rog-core[1430]: [ERROR] Could not get device handle: NoDevice
Jun 07 20:28:58 zg14 rog-core[1430]: Error: NoDevice
Jun 07 20:28:58 zg14 systemd[1]: rog-core.service: Main process exited, code=exited, status=1/FAILURE
Jun 07 20:28:58 zg14 systemd[1]: rog-core.service: Failed with result 'exit-code'.
Jun 07 20:28:58 zg14 systemd[1]: rog-core.service: Scheduled restart job, restart counter is at 5.
Jun 07 20:28:58 zg14 systemd[1]: Stopped ROG Core Daemon.
Jun 07 20:28:58 zg14 systemd[1]: rog-core.service: Start request repeated too quickly.
Jun 07 20:28:58 zg14 systemd[1]: rog-core.service: Failed with result 'exit-code'.
Jun 07 20:28:58 zg14 systemd[1]: Failed to start ROG Core Daemon.
```

I'll keep poking around, and I'm happy to provide any more info that could be useful

dragonnn, 2020-06-09:
https://github.com/flukejones/rog-core/issues/14 you need use animatrix branch

flukejones, 2020-06-09:
I will create a new release `v0.11.0` tonight with these fixes in.

closed 2020-06-09 by flukejones