flukejones/rog-core #22: Add G531GT Support

azagaya, 2020-06-22:
Hi! I own a G531GT. I wanted to know what i need to provide to get it supported. Maybe with some guidance i can add the support myself.

I'm using Ubuntu 20.04, 5.4.0 kernel.

Thanks

flukejones, 2020-06-22:
**Please provide the following:**
 - Distro:
 - Kernel (`uname -r`):
 - `cat /sys/class/dmi/id/product_name`:
 - `cat /sys/class/dmi/id/product_family`:
 - `cat /sys/class/dmi/id/board_name`:
 - `lsusb |grep 0b05`
   + `sudo lsusb -vd 0b05:<all devices from above> > ~/laptop_info`, and attach to issue.

Try running rog-core then:
 - `sudo journalctl -b --unit rog-core > rog-core.log` and attach to issue.
 - `sudo usbhid-dump > usbhid.log` and attach to issue.

flukejones, 2020-06-22:
Additionally, a description of what keyboard LED features exist, if any, such as if it is RGB, multizone, or single colour etc.. A photo of the keyboard too.

azagaya, 2020-06-22:
Distro: Ubuntu 20.04
Kernel: 5.4.0-37-generic
cat /sys/class/dmi/id/product_name : ROG Strix G531GT_G531GT
cat /sys/class/dmi/id/product_family : ROG Strix
cat /sys/class/dmi/id/board_name : G531GT
lsusb | grep 0b05 : Bus 001 Device 002: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device

[laptop_info](https://github.com/flukejones/rog-core/files/4812975/laptop_info.txt)
[rog-core.log](https://github.com/flukejones/rog-core/files/4812981/rog-core.log)
[usbhid.log](https://github.com/flukejones/rog-core/files/4812986/usbhid.log)

Its a RGB keyboard. One Zone RGB lighting. Aura Sync. Hotkeys (Volume down, volume up, mute, HyperFan, Armoury Crate)
[image]

Thanks!

flukejones, 2020-06-23:
Looks very similar to the GX502. Try latest master as of commit 08801f8bfc2b1c55568040f1747b703135547329

Additionally, what is the key next to the the ROG key there? If you do `sudo journalctl -b --unit rog-core > rog-core.log` after pressing it, that will tell us the code it uses, and then we can map it to something.

flukejones, 2020-06-23:
Also, if you can, try each of the modes here `rog-core led-mode --help` and see which ones work, then a profile for your specific laptop can be set.

azagaya, 2020-06-23:
Hi!
That commit works on my notebook. This are the led modes i saw they worked (the rest just turned off the lights):

- stable
- breathe
- strobe (not sure if it does what it should)
- rainbow
- pulse

By the ROG key, i have a key with a fan icon. This is de response of the requested command:

[rog-core.log](https://github.com/flukejones/rog-core/files/4819300/rog-core.log)

azagaya, 2020-06-23:
Also, son functions in the F keys don't trigger any event. For example, Fn+F5 and Fn+F10 (screen of and touchpad off) don't trigger anything. The rest of the functions on those keys seem to work

azagaya, 2020-06-23:
Ok, so i modified the laptops.rs file to include only the led-modes that work.. is that ok?

flukejones, 2020-06-23:
Thanks for the info, I'm adding the correct profile now.

For fanmode you will need a newer kernel, 5.6.10 or newer. The 5.4 kernel doesn't have the required code for the fanmode switching.

Can you try `sudo journalctl -b --unit rog-core > rog-core.log` after pressing the touchpad toggle button? Need to check what the output code is.

For display off, the button does work, but I haven't worked out what the correct command to trigger in Linux actually is.

azagaya, 2020-06-23:
I have the touchpad toogle button in F10 key. Pressing Fn+F10 does not report anything with journalctl.

flukejones, 2020-06-23:
> Also, i cannot see any output when pressing the mute button by the fan button.

Mic mute? Yeah haven't figured out the correct output to trigger mic mute yet. Some keyboards output something like F13-F24...

New commit in master branch. If you can change [block here](https://github.com/flukejones/rog-core/blob/3fec275fa03ab955cf231332a1f43ed4097c2e81/rog-core/src/laptops.rs#L221) to:

```
            GX502Keys::TouchPadToggle => {
                info!("TouchPadToggle: {:X?}", &key_buf);
                let mut key = [0u8; 32];
                key[0] = 0x01;
                key[3] = 0x070;
                rogcore.virt_keys().press(key);
            }
```

we can check the key output.

azagaya, 2020-06-23:
Well, i changed those lines, then `make` and  `sudo make install`, reloaded the daemon. But still nothing gets printed out when i press Fn+F10.

azagaya, 2020-06-23:
I'm sorry, the daemon was dead idk why.. i restarted it and now i got this:

`jun 23 20:46:42 azagaya-G531GT rog-core[6007]: [INFO ] TouchPadToggle: [5A, 6B, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
`

flukejones, 2020-06-24:
The touchpad toggle key is being caught correctly. I'm unsure what the issue is here. When you press that key a virtual keyboard device then emits the correct key, which you should be able to map to something if desired. For me I didn't need to do anything, Gnome just just works with it.

azagaya, 2020-06-24:
When i try to mapp it to another keys in xfce, it appears as Supper+R. Thanks, i think this solves it. But now, the Fn+F7 (turn screen of) doesn't report anything. Can i make similar changes to get it emit the signal? 
If this is no longer relate to rog-core though, we can close this issue. 
Thanks for this cool project!

flukejones, 2020-06-25:
>  But now, the Fn+F7 (turn screen of) doesn't report anything. Can i make similar changes to get it emit the signal?

I've been trying a few different key codes for brightness control as specified by the USB HID standard, but unfortunately none are working. I may change it to something that can be mapped at a later date.

Glad to hear that otherwise, things are working for you.

closed 2020-06-25 by flukejones