flukejones/rog-core #27: ROG Strix G512LU

Pastura, 2020-07-03:
**Describe the bug**
Regardless of distribution, the following FN keys do not work on this laptop:
- brightness keys
- volume key
- AURA keys
- keyboard light keys

The brightness can be changes through Gnome GUI and also by using the `light` CLI tool. 

All led lights work by using this tool: https://github.com/wroberts/rogauracore

rog-core is not working:
```
~ >>> rog-core -d                                                                                                                                                           
INFO: Product name: ROG Strix G512LU_G512LU
INFO: Board name: G512LU
thread 'main' panicked at 'Unsupported laptop, please request support at
```

**To Reproduce**
Buy G512LU
Be sad about Fn keys not working.

**Expected behavior**
Fn keys should work

**Please provide the following:**
 - Distro: Manjaro
 - Kernel (`uname -r`): 5.7.0-3-MANJARO
 - `cat /sys/class/dmi/id/product_name`:ROG Strix G512LU_G512LU
 - `cat /sys/class/dmi/id/product_family`: ROG Strix
 - `cat /sys/class/dmi/id/board_name`:G512LU
 - `lsusb |grep 0b05`: Bus 001 Device 002: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
--> can not attach (github bug)

 - `sudo journalctl -b --unit rog-core` > rog-core.log
[rog-core.log](https://github.com/flukejones/rog-core/files/4869703/rog-core.log)

 - `sudo usbhid-dump > usbhid.log` and attach to issue.
[usbhid.log](https://github.com/flukejones/rog-core/files/4869708/usbhid.log)

**Additional context**
Please let us know if all issues stated are within the scope of this repo.

flukejones, 2020-07-03:
Added support in [v0.14.1](https://github.com/flukejones/rog-core/releases/tag/v0.14.1) release. Can you confirm it behaves as expected?

Does this laptop have per-key RGB effects? I expect the specs I find on ASUS website to match accurately, but just double checking here.

If all works as it should please feel free to close this.

Pastura, 2020-07-04:
Yes, it works exceptionally well now. Thank you! You made our day!

Thank you for also updating the AUR package!

Pastura, 2020-07-04:
After installing, I tested by starting the daemon with `rog-core -d`. I saw that all functions were working, however after that i did `sudo systemctl enable rog-core.service` and rebooted the laptop.

Sadly after reboot, the keyboard stopped working completely. I opened the on screen keyboard from gnome, disabled the service, rebooted again and I have a keyboard now.

With a working keyboard, I figured that my kernel still had some params (`acpi_osi=! acpi_osi='Windows 2009'`) so I removed the params, generated the grub config and rebooted again.

Now I am just using kernel `5.7.0-3-MANJARO` without any special params. the `rog-core` service refuses to start:
```
iul 04 14:50:24 adelina-strix systemd[1]: Started ROG Core Daemon.
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: Version: 0.14.1
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: Product name: ROG Strix G512LU_G512LU
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: Board name: G512LU
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: Supported Keyboard LED modes are:
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: - Static
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: - Breathing
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: - Strobing
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: - Rainbow
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: - Pulse
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: - 4-Zone Static Colours
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: If these modes are incorrect or missing please request support at https://github.com/flukejon>
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: Config loaded
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: INTERVAL: 1
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: MAX_PKT_SIZE: 64
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: SYNC: NoSync
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: TRANSFER_TYPE: Interrupt
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: ENDPOINT: 83
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: INTERFACE: 2
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: RogCore loaded
iul 04 14:50:24 adelina-strix rog-core[2014]: WARN: Fan mode: Fan mode not available
iul 04 14:50:24 adelina-strix rog-core[2014]: INFO: Reloaded battery charge limit
iul 04 14:50:24 adelina-strix rog-core[2014]: WARN: Failed to open battery charge limit path: Os { code: 2, kind: NotFound, message: "No such fil>
iul 04 14:50:24 adelina-strix rog-core[2014]: WARN: Battery charge limit: No such file or directory (os error 2)
iul 04 14:50:24 adelina-strix rog-core[2014]: thread 'main' panicked at 'Mode not convertable to array', rog-client/src/lib.rs:154:18
iul 04 14:50:24 adelina-strix rog-core[2014]: note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
iul 04 14:50:25 adelina-strix systemd[1]: rog-core.service: Main process exited, code=dumped, status=6/ABRT
iul 04 14:50:25 adelina-strix systemd[1]: rog-core.service: Failed with result 'core-dump'.
iul 04 14:50:25 adelina-strix systemd[1]: rog-core.service: Scheduled restart job, restart counter is at 5.
iul 04 14:50:25 adelina-strix systemd[1]: Stopped ROG Core Daemon.
iul 04 14:50:25 adelina-strix systemd[1]: rog-core.service: Start request repeated too quickly.
iul 04 14:50:25 adelina-strix systemd[1]: rog-core.service: Failed with result 'core-dump'.
iul 04 14:50:25 adelina-strix systemd[1]: Failed to start ROG Core Daemon.
```

The daemon also refuses to start:
```
~ >>> rog-core -d                                                                                                                                
INFO: Version: 0.14.1
INFO: Product name: ROG Strix G512LU_G512LU
INFO: Board name: G512LU
INFO: Supported Keyboard LED modes are:
INFO: - Static
INFO: - Breathing
INFO: - Strobing
INFO: - Rainbow
INFO: - Pulse
INFO: - 4-Zone Static Colours
INFO: If these modes are incorrect or missing please request support at https://github.com/flukejones/rog-core
INFO: Config loaded
INFO: INTERVAL: 1
INFO: MAX_PKT_SIZE: 64
INFO: SYNC: NoSync
INFO: TRANSFER_TYPE: Interrupt
INFO: ENDPOINT: 83
INFO: INTERFACE: 2
INFO: RogCore loaded
WARN: Fan mode: Fan mode not available
INFO: Reloaded battery charge limit
WARN: Failed to open battery charge limit path: Os { code: 2, kind: NotFound, message: "No such file or directory" }
WARN: Battery charge limit: No such file or directory (os error 2)
thread 'main' panicked at 'Mode not convertable to array', rog-client/src/lib.rs:154:18
stack backtrace:
   0:     0x55648241c25a - <unknown>
   1:     0x55648239d73c - <unknown>
   2:     0x55648241b754 - <unknown>
   3:     0x55648241b18a - <unknown>
   4:     0x55648241a92a - <unknown>
   5:     0x5564823f3794 - <unknown>
   6:     0x5564823915e8 - <unknown>
   7:     0x55648240a586 - <unknown>
   8:     0x55648242aa16 - <unknown>
   9:     0x556482409f65 - <unknown>
  10:     0x7fa2d65d0002 - __libc_start_main
  11:     0x55648237a0ae - <unknown>
  12:                0x0 - <unknown>
zsh: abort      rog-core -d
~ >>>                        
```

flukejones, 2020-07-04:
Did you try to set Multistatic by any chance?

Pastura, 2020-07-04:
> Did you try to set Multistatic by any chance?

I changed the colors by using Fn+Arrows just before rebooting and it was working. I don't know what setting it had before the reboot.

Do you think a solution might be to gracefully handle errors when parsing the array, instead of triggering panic?

flukejones, 2020-07-04:
No because of so many other parts in play. I see the cause in the detail you provided. I *really* need to work out how to unit test the app without hardware.

To me it looks like the cause is multistatic and that is a mode you can now switch to problematically using the aura keys ( the keyboard LEDs only accept input, we can't tell which mode the keyboard is in unless we record it).

I will push an update now, takes 5-20 minutes for deb. I unfortunately need to hit bed, it's after midnight here.

Pastura, 2020-07-04:
Thank you for the push and I will keep an eye on the ticket and help you as much as I can with troubleshooting.

Thank you for your help so far and have a good night!

Pastura, 2020-07-04:
> I _really_ need to work out how to unit test the app without hardware.

There are some situations where unit testing is sadly not really possible. However you do have a community providing support, and maybe you can do a development branch for situations like this, so that we provide QA for you without having any risks on master/releases.

flukejones, 2020-07-04:
> There are some situations where unit testing is sadly not really possible. However you do have a community providing support, and maybe you can do a development branch for situations like this, so [...]

I try to, the G14 channel on [ZappeLs discord server](https://discord.gg/KTmaeA) seems to be where we all congregate. Normally I don't push an update unless I think it is safe but in this case I misjudged with including changes to how laptop modes got handled.

flukejones, 2020-07-04:
New release v0.14.2 shouldn't crash, and should output some detail of what did cause a crash. That output will be in the log and that is what I'll need to fix this. I'm 99% sure the issue lies in multistatic mode, and if this is preventing you using rogcore you can do `sudo vim /etc/rogcore.conf`, and set `"current_mode": 0,` to allow use until the issue is debugged.

Pastura, 2020-07-04:
v0.14.2 is crashing on the next panic:
```
thread 'main' panicked at 'Mode not convertable to array', rog-client/src/lib.rs:198:18
```

However changing `current_mode` to `0` does allow the daemon to start and gets aura and Fn keys working.

flukejones, 2020-07-05:
If you're building from master can you try again please? I found the actual root cause of the problem (illogical logic :grimacing: ). It *should* be fixed now, once you confirm I will push new release.

If you can paste full logs from `sudo journalctl -b --unit rog-core` that would be helpful too.

flukejones, 2020-07-05:
Turns out I can just change my laptop rog profile to any other and if a mode isn't supported by my keyboard it just ignores it - so I can use this to test the logic.

Oh and I already create a virtual device for keyboard commands, so unit tests can use another virtual device with the same HID profile. :shrug:  problem solved.

I've fixed the bug here and will now do a new release.

flukejones, 2020-07-05:
This is fixed now. Very sorry about any issue that caused for you 
trying to use rogcore.

Pastura, 2020-07-05:
> Oh and I already create a virtual device for keyboard commands, so unit tests can use another virtual device with the same HID profile.  problem solved.

Awesome!

> This is fixed now. Very sorry about any issue that caused for you trying to use rogcore.

No worries, we are glad to help!

Pastura, 2020-07-05:
adelina@adelina-strix Linux 5.7.0-3-MANJARO x86_64 20.0.3 Lysia
~ >>> systemctl status rog-core                                                                                 
● rog-core.service - ROG Core Daemon
     Loaded: loaded (/usr/lib/systemd/system/rog-core.service; enabled; vendor preset: disabled)
     Active: failed (Result: core-dump) since Sun 2020-07-05 18:40:31 EEST; 2min 24s ago
    Process: 1363 ExecStart=/usr/bin/rog-core -d (code=dumped, signal=ABRT)
   Main PID: 1363 (code=dumped, signal=ABRT)

iul 05 18:40:31 adelina-strix systemd[1]: rog-core.service: Scheduled restart job, restart counter is at 5.
iul 05 18:40:31 adelina-strix systemd[1]: Stopped ROG Core Daemon.
iul 05 18:40:31 adelina-strix systemd[1]: rog-core.service: Start request repeated too quickly.
iul 05 18:40:31 adelina-strix systemd[1]: rog-core.service: Failed with result 'core-dump'.
iul 05 18:40:31 adelina-strix systemd[1]: Failed to start ROG Core Daemon.
~ >>> sudo systemctl restart rog-core                                                                        [3]
[sudo] password for adelina: 
~ >>> sudo systemctl status rog-core                                                                            
● rog-core.service - ROG Core Daemon
     Loaded: loaded (/usr/lib/systemd/system/rog-core.service; enabled; vendor preset: disabled)
     Active: failed (Result: core-dump) since Sun 2020-07-05 18:43:59 EEST; 1min 2s ago
    Process: 2230 ExecStart=/usr/bin/rog-core -d (code=dumped, signal=ABRT)
   Main PID: 2230 (code=dumped, signal=ABRT)

iul 05 18:43:59 adelina-strix systemd[1]: rog-core.service: Scheduled restart job, restart counter is at 5.
iul 05 18:43:59 adelina-strix systemd[1]: Stopped ROG Core Daemon.
iul 05 18:43:59 adelina-strix systemd[1]: rog-core.service: Start request repeated too quickly.
iul 05 18:43:59 adelina-strix systemd[1]: rog-core.service: Failed with result 'core-dump'.
iul 05 18:43:59 adelina-strix systemd[1]: Failed to start ROG Core Daemon.
~ >>> rog-core -d                                                                                                            [3]
INFO: Version: 0.14.3
INFO: Product name: ROG Strix G512LU_G512LU
INFO: Board name: G512LU
INFO: Supported Keyboard LED modes are:
INFO: - Static
INFO: - Breathing
INFO: - Strobing
INFO: - Rainbow
INFO: - Pulse
INFO: - 4-Zone Static Colours
INFO: If these modes are incorrect or missing please request support at https://github.com/flukejones/rog-core
INFO: Config loaded
INFO: INTERVAL: 1
INFO: MAX_PKT_SIZE: 64
INFO: SYNC: NoSync
INFO: TRANSFER_TYPE: Interrupt
INFO: ENDPOINT: 83
INFO: INTERFACE: 2
WARN: Could not claim keyboard device interface: InvalidParam
WARN: Trying to detach kernel driver then retry
WARN: Kernel driver not active, continue
ERROR: Could not claim keyboard device interface: InvalidParam
ERROR: Invalid parameter
thread 'main' panicked at 'Invalid parameter', /build/rust/src/rustc-1.44.0-src/src/libstd/macros.rs:16:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
zsh: abort      rog-core -d
~ >>>                                                                                                               ~ >~ >~~~ >~ >>~ >>~~~ >>>                                                                                                                              [134]

flukejones, 2020-07-07:
@Pastura can you try the steps [here](https://github.com/flukejones/rog-core/issues/12#issuecomment-653946255) please? Possibly you might need to change `ACTION=="add",` to `ACTION=="bind",`.

What the log above is telling me is that the kernel driver isn't active yet, so yeah, device not initialised and so nothing to detach, leading to a fail. I have some ideas to try if you're comfortable with building from scratch?

flukejones, 2020-07-07:
Latest master has a couple of changes I'm hoping will help determine how/when the fail occurs.

flukejones, 2020-07-07:
The method of initialization I've settled on is a udev rule to trigger the service. It seems the safest and best way to ensure the service is started safely and not too early in the hardware bringup.

File [here](https://github.com/flukejones/rog-core/blob/master/data/rog-core.rules). This is much the same as what @isushik94 was using, but modified a bit to try catch the few devices rog-core does support (vs just product ID 1866)

I've tested it here and it seems to work flawlessly, and given other feedback I'll push a new version tag now.

Pastura, 2020-07-07:
I bumped the AUR package by using the upgrade script (from which I removed the git commands since there is no need to add/commig). This was the easy way to get the new version and the new sha:
```
~/.../rog-core/rog-core >>> ./update.sh                                                                                              ±[●][master]
0.14.3 -> 0.14.4
~/.../rog-core/rog-core >>> makepkg -si                                                                                              ±[●][master]
==> Making package: rog-core 0.14.4-1 (Ma 07 iul 2020 17:39:25 +0300)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found v0.14.4.tar.gz
==> Validating source files with sha256sums...
    v0.14.4.tar.gz ... Passed
==> Extracting sources...
  -> Extracting v0.14.4.tar.gz with bsdtar
==> Starting build()...
```

However after build, the install crashed:
```
==> Finished making: rog-core 0.14.4-1 (Ma 07 iul 2020 17:41:47 +0300)
==> Installing package rog-core with pacman -U...
[sudo] password for adelina: 
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) rog-core-0.14.4-1

Total Installed Size:  1,09 MiB
Net Upgrade Size:      0,02 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking keys in keyring                                                          [###################################################] 100%
(1/1) checking package integrity                                                        [###################################################] 100%
(1/1) loading package files                                                             [###################################################] 100%
(1/1) checking for file conflicts                                                       [###################################################] 100%
error: failed to commit transaction (conflicting files)
rog-core: /lib exists in filesystem (owned by filesystem)
rog-core: /lib/systemd/system/rog-core.service exists in filesystem
Errors occurred, no packages were upgraded.
==> WARNING: Failed to install built package(s).
```

I also tried to delete the old version and clean install:
```
~/.../rog-core/rog-core >>> sudo pacman -Rsc rog-core                                                                                ±[●][master]
checking dependencies...

Packages (1) rog-core-0.14.3-1

Total Removed Size:  1,07 MiB

:: Do you want to remove these packages? [Y/n] 
:: Processing package changes...
(1/1) removing rog-core                                                                 [###################################################] 100%
:: Running post-transaction hooks...
(1/3) Reloading system manager configuration...
(2/3) Arming ConditionNeedsUpdate...
(3/3) Reloading system bus configuration...
~/.../rog-core/rog-core >>> makepkg -i                                                                                               ±[●][master]
==> WARNING: A package has already been built, installing existing package...
==> Installing package rog-core with pacman -U...
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) rog-core-0.14.4-1

Total Installed Size:  1,09 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking keys in keyring                                                          [###################################################] 100%
(1/1) checking package integrity                                                        [###################################################] 100%
(1/1) loading package files                                                             [###################################################] 100%
(1/1) checking for file conflicts                                                       [###################################################] 100%
error: failed to commit transaction (conflicting files)
rog-core: /lib exists in filesystem (owned by filesystem)
Errors occurred, no packages were upgraded.
==> WARNING: Failed to install built package(s).
```

I have no idea why, but it seems to be unhappy about the fact that `/lib/` already exists on the system.

What I can do is test `/etc/udev/rules.d/99-rogcore.rules` with `0.14.3` till you check out file ownership under pacman.

isushik94, 2020-07-07:
> I have no idea why, but it seems to be unhappy about the fact that /lib/ already exists on the system.

@Pastura, Installation paths were changed in a Makefile and it puts some files into `/lib` but not into `/usr/lib`. I have no idea why it has been done. You can edit PKGBUILD to move the package's `lib` into 'usr/`.

flukejones, 2020-07-07:
File locations were changed to match debian and fedora guidelines sorry

Pastura, 2020-07-09:
All issues fixed on Manjaro with version `0.15.0-2` from AUR.

Congrats on the team for the hard work!

closed 2020-07-09 by Pastura