flukejones/rog-core #29: G14

N214, 2020-07-12:
After installing rog-core, it freezes my keyboard on the login screen... can't type my password.

isushik94, 2020-07-12:
Hi
Have you installed 0.15.0 release? It should work (Arch or Ubuntu). But I have G15 laptop. Maybe you need some kernel patches.

P.S. You will need an external keyboard to manage this issue

N214, 2020-07-12:
Hi
it was the 0.14 release but I'll try again now. And I tried to install the custom kernel from AUR but it didn't compile...
An external keyboard can circumvent the issue but not solve it

isushik94, 2020-07-12:
:D
Sorry, I've meant "to help manage the issue"
Yes, try 0.15 as it has changes in a udev rule, which actually starts the service.

isushik94, 2020-07-12:
> And I tried to install the custom kernel from AUR but it didn't compile...

Try install from [here](https://drive.google.com/drive/folders/1BSdHshprfi-QkcH-0ma__MQ1aIi91Wr-?usp=sharing)

N214, 2020-07-12:
Thanks but
I can't start at all, systemd shows: rog-core.service: Start request repeated too quickly.
And if I try to start rog-core directly with rog-core -d it shows: "thread 'main' panicked at 'Could not deserialise /etc/rogcore.conf', rog-core/src/config.rs:49:41"

isushik94, 2020-07-12:
Try delete `/etc/rogcore.conf`. I believe there were some changes in its format

N214, 2020-07-12:
> > And I tried to install the custom kernel from AUR but it didn't compile...
> 
> @N214, try install from [here](https://drive.google.com/drive/folders/1BSdHshprfi-QkcH-0ma__MQ1aIi91Wr-?usp=sharing)

Thanks, that's it, it work fine now. However, I'm getting some systemd warnings like

```
Jul 12 16:18:52 n214-nico rog-core[763]: WARN: Fan mode: Fan mode not available
Jul 12 16:18:52 n214-nico rog-core[763]: INFO: Reloaded battery charge limit
Jul 12 16:18:52 n214-nico rog-core[763]: WARN: Failed to open battery charge limit path: Os { code: 2, kind: NotFound, message: "No such file or directory" }
Jul 12 16:18:52 n214-nico rog-core[763]: WARN: Battery charge limit: No such file or directory (os error 2)
```

isushik94, 2020-07-12:
Run `lsmod | grep asus` and check for `asus_nb_wmi`. If it's not in the list, check the kernel you have loaded with `uname -a`. There must be a part looking like this `5.7.6-arch1-1-g14`.

N214, 2020-07-12:
Yes I have `asus_nb_wmi`
Here is the output
`asus_nb_wmi            28672  0`

isushik94, 2020-07-12:
Looks good...
I don't know how to help with it, so it's better to wait @flukejones
Meanwhile check this website https://lab.retarded.farm/zappel/asus-rog-zephyrus-g14

N214, 2020-07-12:
Thanks, I solved the issue. 
Everything works now, I tried all the fans mode and it seems that "normal" is the lowest and "silent" and "boost" are pretty much the same, is it normal?

isushik94, 2020-07-12:
The "silent" must use coolers less but it controlled by asus software. And you can edit `/etc/rog-core.conf` to disable turbo for instance

N214, 2020-07-12:
Sorry didn't get what you said what is "coolers less" and controlled by asus sofware? 
So currently the best we can do is to set to "normal mode" and "no_turbo" to true

isushik94, 2020-07-12:
It's about a laptop's cooling system. When "silent" is on, it must be turned off, roughly speaking. It's noticeable when laptop does heavy calculations.
And "asus software" is a bios or driver which controls it

N214, 2020-07-12:
Ah ok thanks for the explanation so currently the best we can do is having the fans spinning between 2600 and 3500rpm? It's quite loud... At least on my G14

dragonnn, 2020-07-12:
Nope, silent with disabled boost is about 2200-2400RPM and is quite stable. When using only the iGPU of course.

N214, 2020-07-12:
> Nope, silent with disabled boost is about 2200-2400RPM and is quite stable. When using only the iGPU of course.

How to do that? I'm on my iGPU, turbo disabled, it still ranges from 2900-3500RPM

dragonnn, 2020-07-12:
Maybe something in you system is loading it to high? 
```
uptime                                                                                                                                                                                                                      
 16:25:15 up  3:10,  1 user,  load average: 1,06, 0,81, 0,79
```
my load average

N214, 2020-07-12:
Seems so and how can I troubleshoot that?
` 16:26:08 up  1:50,  1 user,  load average: 1.01, 1.27, 1.35
`

dragonnn, 2020-07-12:
Well that deepens on you DE and installed programs. You just need to look at process sorted by CPU usage and look what is spiking from time to time at idle. But I think this issue should be closed since this has nothing to do with rog-core, you can ask for help on our discord https://discord.gg/CEm2ZA

N214, 2020-07-12:
Yes thanks to both !

closed 2020-07-12 by N214