flukejones/rog-core #30: G531GW more supportred led-modes

kunver400, 2020-07-20:
@devs  great work, this exactly what i was looking for.

I Edited the laptops.rs and found these effects to be working

Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Static // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Breathing // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Strobing // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Rainbow // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Comet // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Flash // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Keypress Highlight // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Rain // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Keypress Ripple // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Stars // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - RGB per-key // Not Working yet
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - Pulse // Working
Jul 20 15:41:23 V3X-ARCH rog-core[462]: INFO: - 4-Zone Static Colours // Not supported on this machine 

`[SINGLE, BREATHING, STROBE, RAINBOW, COMET, FLASH, HIGHLIGHT, RAIN, RIPPLE, STAR, PULSE]`
Can these be merged?

flukejones, 2020-07-20:
Looks like gx502 level. Will add soon.

closed 2020-07-21 by flukejones

kunver400, 2020-07-21:
@flukejones this is not gx531 it's G531(G531GW-AZ051T).

flukejones, 2020-07-21:
Ah bugger, sorry mate it was late here, I must have crossed wires when reading.

There doesn't really seem to be any rhyme or reason behind the model numbering either, so when looking at GX531, it seems to have multi-zone only while the G531 has per-key and zones...

I'll correct this now and check the other laptops again.

kunver400, 2020-07-21:
Totally agree, FYI this is a Rog strix scar 3 (G531GW), ASUS ROG Strix Hero III (G531GU), Asus ROG Strix G (GL531GT) are likely to have the same config.

commit 4ba7a37e (2020-07-21): Prep new release

Closes #30, #31, #32
commit dd897315 (2020-07-21): Prep new release

Closes #30, #31, #32
commit bffe40d7 (2020-07-21): Prep new release

Closes #30, #31, #32