flukejones/rog-core #31: G531GW Per key RGB, Light bars, Logo

kunver400, 2020-07-20:
Hi,

I noticed that g531 is listed under 4-zone RGB, It supports per key and is controlled via aura creator on windows.
switching to 4-zone rgb just turns off the lights altogether.

I am not sure if it is possible to have per key controls, but i can provide wireshark captures.
Also it will be nice to have some control over these light bars and logo on the lid.
[image]

flukejones, 2020-07-20:
If it's like gx502 then the lights under laptop are part of keyboard EC. Lid is not likely to be controllable (default red?). Per-key will be exactly same as gx502. Will add this and other modes later.

kunver400, 2020-07-20:
Lid logo is also RGB,  Light bars do look like a part of keyboard EC.

flukejones, 2020-07-21:
You might need to do a single wireshark capture of the lid LED to verify how/where it is set. I suspect it will be part of the per-key byte arrays.

kunver400, 2020-07-21:
Armoury crate only has options to toggle lightzones (i.e bottom lights and  lid logo together), i have used aura creator to turn on light bars and logo individually while having the rest of the lights off.

I have attached captures for both.

[g531gw_captures.zip](https://github.com/flukejones/rog-core/files/4951467/g531gw_captures.zip)

flukejones, 2020-07-21:
Notes:

Light-zones seems to be a new mode? `5dbd010c050000000...`

Light-bars is a per-key function, appears to be an addition to the protocol so there are 12 sets of interrupt writes compared to the GX502 (11 writes). Look for the **ff**, which if RED led was used means these are the start of RGB, the two `00` following the `ff` will be G and B:

```
5dbc00010400000000000000000000ff0000ff0000ff0000ff0000ff0000ff000000000000000000000000000000000000000000000000000000000000000000
```

Looks like 6 LEDs for the light bars, each individually controlled. @kunver400 can you map these? Do one capture per LED. I used aura creator to set one, then run wireshark, and press the play/test button, then stop capture. **Use white colour**

Lid logo seems to be in first write of loop as: 

```
5dbc00010101001000ff000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

@kunver400 how janky are the effects you set? The protocol for RGB really is kind of bad, the ideal is that each interrupt write takes 1ms or less, but often can take up to 2ms, and in windows one write takes up to 45ms on average. So a full write for every key for you ends up being worst case scenario 12 (writes) * 2ms = 44ms + ~40ms.. It is actually a little better in Linux and will average around 16ms to 25ms. Just means the frame rate is very low, or you notice a lot of keys getting missed at higher speeds.

flukejones, 2020-07-21:
Given the above it won't be too hard to implement the missing bits, but I'd rather like to have a good think about how best to do this.

If you want to test the existing RGB stuff, try `cargo run --example ball`

kunver400, 2020-07-21:
[image]

yes, these are the six leds,

```
5dbc00010400000000000000000000000000000000000000000000000000ffffff00..	LED1
5dbc00010400000000000000000000000000000000000000000000ffffff00..	LED2
5dbc00010400000000000000000000000000000000000000ffffff00..	LED3
5dbc00010400000000000000000000000000000000ffffff00..	LED4
5dbc00010400000000000000000000000000ffffff00..	LED5
5dbc00010400000000000000000000ffffff00..	LED6
```

it seems like keyboard layout is a little different from gx502, while running iteratekeys some keys where not lighting up, namely:
ROGKEY, F6, TAB, ENTER, RSHIFT, FN

i have captured these individually and attached along with the led captures.

[morecaptures.zip](https://github.com/flukejones/rog-core/files/4953329/morecaptures.zip)

kunver400, 2020-07-21:
Lightzones look pointless as per key controls already have these, ASUS must have added this mode to turn off lightbars in builtin effects.

kunver400, 2020-07-21:
@flukejones Let me know if you need individual key captures.

kunver400, 2020-07-21:
Another observation : only ROGKEY, ENTER, RSHIFT, are not lighting up, when testing with the pulser example. rest of the keys are looking good.

flukejones, 2020-07-21:
> Lightzones look pointless as per key controls already have these, ASUS must have added this mode to turn off lightbars in builtin effects.

If you can get a capture per zone I would really appreciate it. This is mostly so I can check things and for completeness.

flukejones, 2020-07-21:
Pretty simple to insert the extra keys - the spreadsheet [here](https://github.com/flukejones/rog-core/blob/master/reverse_eng/per_key_raw_bytes.ods) shows where they end up in the protocol. I'll add them to rog-core properly later when I decide on how to refactor the per-key architecture.

kunver400, 2020-07-21:
@flukejones there are no multiple lightzones , this is just a single checkbox in armoury crate, toggling this turns off everything except for the keyboard. i have used white color but i think this is just a switch.
[image]

captures: [Lightzones-toggle.zip](https://github.com/flukejones/rog-core/files/4957017/Lightzones-toggle.zip)

kunver400, 2020-07-22:
also, looking at the spreadsheet,  we missed the logo on the lid 

`5dbc00010101001000ffffff00..	LID-ROGLOGO`

[G531-LID-LOGO.zip](https://github.com/flukejones/rog-core/files/4957569/G531-LID-LOGO.zip)

flukejones, 2020-07-28:
This is going to be put on the backlog for a week or two. I hope that's okay for now.