flukejones/rog-core #34: Unable to install or build on Ubuntu 18.04

hvico, 2020-07-27:
Hello,

I couldn't install rog-core un my Zephyrus GX501 running Ubuntu 18.04. The provided deb repository seems to host only builds for Ubuntu 20.04 (it gives dependency problems related with libc versions).

Tried to build the project but it fails near the end:

`   Compiling filetime v0.2.10
   Compiling xattr v0.2.2
   Compiling regex v1.3.9
   Compiling mio v0.6.22
   Compiling tar v0.4.29
   Compiling cexpr v0.4.0
   Compiling env_logger v0.7.1
   Compiling syn-mid v0.5.0
   Compiling synstructure v0.12.4
   Compiling libusb1-sys v0.3.7
   Compiling rusb v0.6.0
   Compiling uhidrs-sys v1.0.1
error: failed to run custom build command for `uhidrs-sys v1.0.1`

Caused by:
  process didn't exit successfully: `/home/hvico/rog-core/target/release/build/uhidrs-sys-43b328dadac05f2a/build-script-build` (exit code: 101)
--- stderr
/usr/include/x86_64-linux-gnu/sys/types.h:145:10: fatal error: 'stddef.h' file not found
/usr/include/x86_64-linux-gnu/sys/types.h:145:10: fatal error: 'stddef.h' file not found, err: true
thread 'main' panicked at 'Unable to generate bindings: ()', src/libcore/result.rs:1188:5
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
`

I have latest libclang-dev installed.

cakesale, 2020-07-27:
20.04 is latest LTS version, any reason you're not upgrading?

hvico, 2020-07-27:
Yes, 18.04 is supported until 2023 and I would say is still the de-facto standard for Machine Learning tasks. I will eventually upgrade but not right now. Maybe some advice on where I can specify the include dir for that build task that fails near the end (uhiddrs). It seems it is just a path issue. Thanks

aspann, 2020-07-27:
Hey, what version of clang is installed? (not the libs)

it might be fixed by:
`sudo apt-get install clang-9` (or -10, I'm a bit unsure atm.)

cheers

hvico, 2020-07-27:
Thank you very much! That did the trick:

sudo apt-get install clang-9 libclang-cpp9 libclang-common-9-dev libclang1-9

Now the build works. Thanks.

aspann, 2020-07-27:
You're always welcome! ;)

closed 2020-07-27 by hvico