flukejones/rog-core #35: Please add support for ROG Strix G731GV

TabPaD, 2020-08-01:
Well honestly when I decided to install linux on my machine I was really prepared to give up all the extra functions which was offered with asus drivers on Windows. But then, when I found this project incidently, I was so amazed that I can't even begin to describe...
My Asus ROG G731GV wasn't on the list but I decided to give it a try and the results are quite satisfying.
In fact almost everything other than Fan Modes and LED Modes work fine. Changing Fan modes or LED color either using physical keys or bash commands is not possible and right now I'm stuck with static red color on the keyboard (before installing rog-core it was turned off).
Media and FN+ keys including Mic mute, volume control, screen brightness, touchpad toggle, sleep, airplane mode and even Charge Limit all work fine.

- Distro: Linux Mint 20 Ulyana
- Kernel: 5.4.0-42-generic
- `cat /sys/class/dmi/id/product_name`: ROG Strix G731GV_G731GV
- `cat /sys/class/dmi/id/product_family`: ROG Strix
- `cat /sys/class/dmi/id/board_name`: G731GV
- `lsusb | grep 0b05`: Bus 001 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
- [ ] [laptop_info.LOG](https://github.com/flukejones/rog-core/files/5009614/laptop_info.LOG)

- `sudo journalctl -b --unit rog-core > rog-core.log`
- [ ] [rog-core.log](https://github.com/flukejones/rog-core/files/5009674/rog-core.log)
- `sudo usbhid-dump > usbhid.log
- [ ] [usbhid.log](https://github.com/flukejones/rog-core/files/5009752/usbhid.log)

here are some pictures of the keyboard:
[image]
[image]

flukejones, 2020-08-01:
Are you able to list which LED modes should work for your machine? If you still have windows installed you can cycle through them. Otherwise I'll add your laptop to a default minimal list.

DOes it support RGB per key?

flukejones, 2020-08-01:
Hmm, per-key, 4-zone, and others? Looks very similar to another laptop recently added, I'll follow that one (G531GW)

TabPaD, 2020-08-01:
It has 4 zones. Here is a picture of aura creator app which shows all the lights on the keyboard and sides of laptop.
[image]

Using keyboard shortcuts I can cycle through Static, Breathing, Strobing, Color Cycle and Rainbow. A few other modes like Stary night, Music, Smart and Adaptive color are also available through Aura Sync.

flukejones, 2020-08-02:
If you're comfortable with building the app on your own, the latest master should have the support you need.

TabPaD, 2020-08-02:
I would love to help but unfortunately I'm not very familiar with Rust.

flukejones, 2020-08-02:
Which distro mate?

TabPaD, 2020-08-02:
If you mean which distro I’m using, I’ve mentioned it in my first post. Linux Mint 20 Ulyana

TabPaD, 2020-08-05:
Just a little reminder!!
G731 or at least G731GV has 4-zone RGB and is added in the wrong section.

flukejones, 2020-08-05:
[tech specs](https://www.asus.com/Laptops/ROG-Strix-SCAR-III/Tech-Specs) for reference.

TabPaD, 2020-08-05:
Okay it seems that there are different Versions of G731. e.g. Strix Hero III G731, Strix Scar III G731 and Strix G731.
These are the [correct technical specs](https://www.asus.com/Laptops/ROG-Strix-G/Tech-Specs/) for my laptop!

> One Zone RGB lighting (GT/GU)
> 4 Zone RGB lighting (GV/GW)

This is also shown in the screen shot from Aura Creator app on Windows!

flukejones, 2020-08-10:
Thanks for the detail. I will add these variants at a later date. If you want to do this now it's not a hard task - add a block like this:

```
    } else if board_name.starts_with("G731GV")
        || board_name.starts_with("G731GW")
    {
        laptop.supported_modes = vec![ SINGLE, BREATHING, STROBE, RAINBOW,  MULTISTATIC, ];
    } else if board_name.starts_with("G731GT")
        || board_name.starts_with("G731GU")
    {
        laptop.supported_modes = vec![ SINGLE, BREATHING, STROBE, RAINBOW,  RGB, ];
```

above [this line](https://github.com/flukejones/rog-core/blob/4c32901467fb2b21fc7c6a7243e18e16e36392b8/rog-core/src/laptops.rs#L126).

Add whichever modes are required. Do note that `RGB` here means "per-key RGB". Sorry, it's a bit badly named as this once targeted only one laptop.