flukejones/rog-core #36: Silent fan profile does not persist on GA401

schaerfo, 2020-08-02:
After applying the 'silent' fan profile with
```
$ rog-core -f silent
```
the fan spin down, but only for a few seconds. After that, they return to their normal speed.
In order to have the fans in silent mode continuously, the command above has to be run periodically.

Yarn, 2020-08-02:
The "normal" speeds they return to are still the silent profile (and should be a bit lower than the normal profile). I have custom fan curves working if you want to beta test that, I don't have the instructions anywhere but on the discord (linked in the readme) atm though.

schaerfo, 2020-08-03:
OK, that is some rather unintuitive behavior then. I will perhaps take a look at the custom fan curves, thanks!

closed 2020-08-03 by schaerfo