flukejones/rog-core #37: Problem with rog-core service after suspend (ROG Strix G531GT)

Dusko-Vrljanovic, 2020-08-05:
Hi. I have issue with rog-core service after I bring back laptop from suspend.
I have ROG Strix G531GT, with Manjaro installed and 5.7.9-1 kernel version.

When i did clean install, Fn+ F5 - F12 and Fn + arrows didn't work. I installed rog-core from AUR, and it resolved most of buttons. Only two I noticed that don't work are fan speed control and turn of display. Here are my issues:

- When I press fan speed control, in service status it says: WARN: Couldn't toggle fan mode: Custom { kind: NotFound, error: "Fan mode not available" }

- Chaning fan mode with `rog-core --fan-mode` results in WARN: Custom { kind: NotFound, error: "Fan mode not available" }

- Changing battery charge with `rog-core --charge-limit 80` results in: 
        WARN: Failed to open battery charge limit path: Os { code: 2, kind: NotFound, message: "No such file>
        WARN: Os { code: 2, kind: NotFound, message: "No such file or directory" }

- My biggest problem is, when laptop goes to suspend, and turns on againg, rog-core stops working. 
     Service status is ERROR: Failed to read keyboard interrupt: Io
  In journalctl it logs "usb 1-8: usbfs: process 898 (tokio-runtime-w) did not claim interface 2 before use" a few thousand times a second, and it keeps one thread to 100%. Process with pid 898 is rog-core.

flukejones, 2020-08-05:
Please provide all the information requested in the [bug template](https://github.com/flukejones/rog-core/blob/master/.github/ISSUE_TEMPLATE/bug_report.md).

Please also do the following and attach here:

```
sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
iasl -d dsdt.dat
mv dsdt.dsl (cat /sys/class/dmi/id/board_name).dsl
rm dsdt.dat
```

A kernel patch is waiting to be accepted for mainline, but rog-core is an interim solution and to fix issues I need much more detail other than "it doesn't work". Usually the fan and battery warning there indicates that the kernel didn't set up those devices for example `/sys/devices/platform/asus-nb-wmi/throttle_thermal_policy` should exist.

Dusko-Vrljanovic, 2020-08-05:
- Distro: Manjaro Linux 20.0.3 Lysia

- Kernel: 5.7.9-1-MANJARO

- Product name: ROG Strix G531GT_G531GT

- Product family: ROG Strix

- Board name: G531GT

- `lsusb | grep 0b05`: Bus 001 Device 004: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device

[laptop_info.txt](https://github.com/flukejones/rog-core/files/5028084/laptop_info.txt)

- `sudo journalctl -b --unit rog-core > rog-core.log`
[rog-core.log](https://github.com/flukejones/rog-core/files/5028096/rog-core.log)
I couldn't upload whole .log file, it was almost 60Mb, instead of 3 dots there was almost 600000 lines of "rog-core[892]: ERROR: Failed to read keyboard interrupt: Io"

- `sudo usbhid-dump > usbhid.log` generates empty file, this is error output
001:004:002:Failed to claim: Resource busy
001:004:002:Failed to release: Entity not found
001:004:001:Failed to release: Entity not found
001:004:000:Failed to release: Entity not found
001:003:002:Failed to release: Entity not found
001:003:001:Failed to release: Entity not found
001:003:000:Failed to release: Entity not found
001:002:000:Failed to release: Entity not found

- [G531GT.dsl.txt](https://github.com/flukejones/rog-core/files/5028118/G531GT.dsl.txt)

I do realize that You need more details, and I will provide everything You want, I just didn't know at first everything that you need.

There is no `/sys/devices/platform/asus-nb-wmi/throttle_thermal_policy`, but battery and fan are not that important. Crash after suspend is more important.

flukejones, 2020-08-05:
In absolute honesty, Arch and Arch based distros have been a pain for this. `rog-core` does a couple of janky things to make stuff work and it's not the best - and will be seeing very minimal updates in favour of the new asus-nb-ctrl forked from this (and maintained by me).

I suggest using the kernel [here](https://aur.archlinux.org/packages/linux-g14/), and preferably that you download and customise the build to add the kernel [patch from here](https://gitlab.com/asus-linux/asus-nb-ctrl/-/blob/next/kernel-patch/0001-HID-asus-add-support-for-ASUS-N-Key-keyboard-v5.7.patch). The maintainer of that should be updating the pkgbuild with the patch soon.

Once you have the patched kernel running, install [asus-nb-ctrl](https://gitlab.com/asus-linux/asus-nb-ctrl/-/tree/next). Do please join our discord server.

ngoquang2708, 2020-08-11:
I have the same laptop model, run the same OS. Everything seem working fine for me. It is strange.

flukejones, 2020-08-12:
@Dusko-Vrljanovic please follow instructions [here](https://github.com/flukejones/rog-core/issues/40#issuecomment-672295529)

Dusko-Vrljanovic, 2020-08-13:
@ngoquang2708 it really is. I guess there's always that one off with problems.
@flukejones I tried it, installed g14, asus-nb-ctrl, and added g14 section to conf file, it's still the same.

flukejones, 2020-08-24:
rog-core is now obsoleted, please use https://gitlab.com/asus-linux/asus-nb-ctrl

closed 2020-08-24 by flukejones