flukejones/rog-core #38: Add suport for FA506IV

lvalladares, 2020-08-08:
Hey!

Here is the output of the commands you ask

**lsusb |grep 0b05**
EMPTY

**sudo usbhid-dump**
No matching HID interfaces

**cat /sys/class/dmi/id/product_name**
TUF Gaming FA506IV_TUF506IV

**cat /sys/class/dmi/id/product_family**
TUF Gaming FA506IV

**cat /sys/class/dmi/id/board_name**
FA506IV

I think this output is´nt helpful. Let me know if i can run any other command.

flukejones, 2020-08-10:
rog-core won't ever be able to support the keyboard on this laptop (or any of the TUF series) because the keyboard is connected over a different bus. You will be able to use fan and battery control though (at which point I suggest using [asus-nb-ctrl](https://gitlab.com/asus-linux/asus-nb-ctrl)).

What exactly isn't working for you using the standard kernel?

lvalladares, 2020-08-10:
Right now i'm using 5.7 Kernel and i cant change keyboard light, see or change fan speed, see or change battery limit (but i think its writed at kernel level, because change made in windows with armoury crate do take effect in linux)

For the light i've instaled [Faustus](https://github.com/hackbnw/faustus) and works nice. I'm waiting for kernel 4.8 to be moved from mainline to release to test if it fix something.

I've tried rog-core but i believe it checks for laptop model before trying to do anything because nothing seem to work and when i try to start the daemon it says my laptop its not compatible.

flukejones, 2020-08-10:
Faustus is what you should definitely be using. I will close this for now, a future release of this or asus-nb-ctrl will enable control of fan and battery charge limit *if* the kernel has support.

closed 2020-08-10 by flukejones