flukejones/rog-core #40: Please add support for Strix GL504GV

lrissman-github, 2020-08-11:
**Describe the bug**
Asus Strix GL504GV not supported. 

**To Reproduce**
Install the rog-core package from AUR using pacman
run rog-core -d

see this output
```
INFO: Version: 0.15.0
INFO: Product name: Strix GL504GV
INFO: Board name: GL504GV
thread 'main' panicked at 'Unsupported laptop, please request support at https://github.com/flukejones/rog-core', rog-core/src/laptops.rs:95:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Aborted (core dumped)
```

**Expected behavior**
GL504GV is supported and rog-core functions normally

**Please provide the following:**
 - Distro:
 - Kernel (`uname -r`):
'5.7.14-1-MANJARO'
 - `cat /sys/class/dmi/id/product_name`:
`Strix GL504GV`
 - `cat /sys/class/dmi/id/product_family`:
`Strix`
 - `cat /sys/class/dmi/id/board_name`:
`GL504GV`
 - `lsusb |grep 0b05`
`Bus 001 Device 004: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device`
   + `sudo lsusb -vd 0b05:<all devices from above> > ~/laptop_info`, and attach to issue.
`[laptop_info.txt](https://github.com/flukejones/rog-core/files/5059317/laptop_info.txt)`
 - `sudo journalctl -b --unit rog-core > rog-core.log` and attach to issue.
[rog-core.log](https://github.com/flukejones/rog-core/files/5059324/rog-core.log)
 - `sudo usbhid-dump > usbhid.log` and attach to issue.
[usbhid.log](https://github.com/flukejones/rog-core/files/5059325/usbhid.log)

flukejones, 2020-08-11:
Since you are using Arch please use the `g14` kernel and `asus-nb-ctrl` from the follwing:

 Edit `/etc/pacman.conf` to add

```
[g14]
SigLevel = Optional TrustAll
Server = https://arch.retarded.farm
```

it includes the kernel patch that came from all the rog-core work, and asus-nb-ctrl is the next evolution of rog-core to remove the parts that were pushed to kernel.

lrissman-github, 2020-08-11:
Thank you.  It worked. 

Because I was using the nonfree Nvidia drivers and was on the default 5.7 kernal that comes with Manjaro, I decided to use the tools to update to the included 5.8 and update the nvidia drivers before installing the G14 kernel. 

I can confirm that the function keys, brightness and touchpad controls work properly now including fan control and keyboard backlight brightness.

closed 2020-08-12 by flukejones

acujl, 2020-08-13:
I'm having this exact same issue on G531GU, I'm running PopOS.

- Kernel: 5.4.0-7634-generic
- cat /sys/class/dmi/id/product_name
ROG Strix G531GU_G531GU
- cat /sys/class/dmi/id/product_family
ROG Strix
- cat /sys/class/dmi/id/board_name
G531GU
-lsusb |grep 0b05
Bus 001 Device 003: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device

flukejones, 2020-08-13:
@acujl there are a lot of changes happening. `rog-core` will be archived soon to be replaced with a dkms usable [kernel module](https://gitlab.com/asus-linux/hid-asus-rog) and [asus-nb-ctrl](https://gitlab.com/asus-linux/asus-nb-ctrl). I will try to get packages for Ubuntu/Pop done soon.

Join our discord if you can.

lrissman-github, 2020-08-13:
As an update to you, after adding the G14 kernel and the asus-nb-ctrl, all the hotkeys and everything worked well; however, nothing I did could get the nonfree nvidia drivers to actually work.  the system would see the nvidia with the nonfree driver, but I could not get it to properly offload.  

The nvidia drivers in the g14 repo didnt seem to allow prime-run or the environment variables to offload in any of the attempts i tried.    The nonfree in the manjaro repos wouldnt work with the kernel in the g14/asus-nb-ctrl, and also i couldnt get it to work with the binaries downloaded directly from nvidia.   In all cases inxi -G would show the nvidia properly, but the offload wouldnt work.

After several full reinstalls, i actually settled on this: 
- base manjaro install
- upgraded to manjaro linux58
- added G14 repo
- installed rog-core from G14
- used a manjaro guide to install the non-free drivers. 

So my currently working state is: Manjaro linux58 kernel /w manjaro nonfree nvidia /w rog-core [g14]

Thanks again for this work!  I can tell you, I truly appreciate having my display brightness working.. :+1: